ros2/ros2_documentation #870: Documentation on install ros2-base from source

mturnshek, 2020-09-17:
## Feature request
Instructions for installing ros2-base from source.

#### Feature description
Documentation with a list of the essential packages for each distribution. This would allow a user to install precisely the packages they want, from source, by ignoring all auxiliary packages. This is important, because currently installing from source requires many gigabytes, but not all ros2 packages are used by everyone.

#### Implementation considerations
Alternatively, instructions for how to compile additional packages with new or modified code on top of an existing debian package installation.

gavanderhoorn, 2022-07-06:
Part of this (a bit of explanation at least), seems to have been added by https://github.com/ros2/ros2_documentation/pull/1459.

Explaining the actual *use* of `rosinstall_generator` with these variants is still missing I believe.

Thing is: instructions would largely overlap with the current from-source builds / development environment setup guides.

Would we want a section in `Installation » Alternatives » Ubuntu (source)`, next to [Get ROS 2 code](http://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html#id6), or a new page?

Content would be similar to [this comment in *Core, base and desktop releases* on ROS Discourse](https://discourse.ros.org/t/core-base-and-desktop-releases/20267/2).

Personally, I would perhaps prefer a *Building a ROS 2 installation variant from source* page or something, to not complicate the -- already quite long -- installation from-source page even more, but it would mean adding yet another page which explains how to build ROS 2 from source.

clalancette, 2022-07-06:
> Thing is: instructions would largely overlap with the current from-source builds / development environment setup guides (it would be a guide about using `rosinstall_generator` to build a ROS 2 variant, not a user manual for `rosinstall_generator`).

Yes, one of the reasons we haven't written this tutorial yet is exactly because of this overlap.

We have a long-term goal to stop using [ros2.repos](https://github.com/ros2/ros2/blob/master/ros2.repos) completely and switch to using `rosinstall_generator` commands, but that has a number of implications both for core CI and for downstream consumers.  This is not likely to happen anytime soon, so we can proceed with adding some documentation somewhere for using `rosinstall_generator` specifically for ROS 2.

> Would we want a section in `Installation » Alternatives » Ubuntu (source)`, next to [Get ROS 2 code](http://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html#id6), or a [...]

I don't think we should add a new page; there are already a bewildering array of options.  Instead, what I think we should do is revamp the current https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html page (along with the RHEL, Windows, etc pages) to streamline the current content.  I feel like a bunch of that can be moved elsewhere.  Once we've done that, I think it would be reasonable to add a small section about using `rosinstall_generator`.

gavanderhoorn, 2022-07-06:
So that basically makes it impossible to move forward with this issue, as it's blocked by:

> what I think we should do is revamp the current [docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html](https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html) page (along with the RHEL, Windows, etc pages) to streamline the current content. I feel like a bunch of that can be moved elsewhere.

clalancette, 2022-07-06:
> So that basically makes it impossible to move forward with this issue, as it's blocked by:

It doesn't feel impossible; that revamp seems fairly straightforward to me.