ros2/ros2_documentation #879: ros2 bag installation failed

hsyis, 2020-09-26:
I experienced an installation failure while following this tutorial [Recording and playing back data](https://index.ros.org/doc/ros2/Tutorials/Ros2bag/Recording-And-Playing-Back-Data/#ros2bag).

I used the ros2-eloquent on Ubuntu 18.04.5.
Below is the execution result.
```bash
$ sudo apt-get install ros-eloquent-ros2bag ros-eloquent-rosbag2*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'ros-eloquent-rosbag2-storage-default-plugins-dbgsym' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-test-common' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-converter-default-plugins' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-tests' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-dbgsym' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-bag-v2-plugins' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-storage-dbgsym' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-storage' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-storage-default-plugins' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-converter-default-plugins-dbgsym' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-transport' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-bag-v2-plugins-dbgsym' for glob 'ros-eloquent-rosbag2*'
Note, selecting 'ros-eloquent-rosbag2-transport-dbgsym' for glob 'ros-eloquent-rosbag2*'
ros-eloquent-rosbag2 is already the newest version (0.2.4-1bionic.20200606.033625).
ros-eloquent-rosbag2 set to manually installed.
ros-eloquent-rosbag2-storage is already the newest version (0.2.4-1bionic.20200606.021326).
ros-eloquent-rosbag2-storage set to manually installed.
ros-eloquent-rosbag2-transport is already the newest version (0.2.4-1bionic.20200606.034308).
ros-eloquent-rosbag2-transport set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-eloquent-rosbag2-bag-v2-plugins : Depends: ros-eloquent-ros1-bridge but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

clalancette, 2020-10-15:
I'm going to go ahead and close this with no response for now.  If you are still having problems, I encourage you to create a post at https://answers.ros.org , where you may be able to get more help.

hsyis, 2020-10-15:
@clalancette Oh, I was missing this issue. Sorry for the late response.
Below is the result of executing the command. I'll find out more. Thanks.

```bash
$ sudo apt-get install ros-eloquent-ros1-bridge
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-eloquent-ros1-bridge : Depends: ros-melodic-actionlib-msgs but it is not installable
                            Depends: ros-melodic-catkin but it is not installable
                            Depends: ros-melodic-common-msgs but it is not installable
                            Depends: ros-melodic-gazebo-msgs but it is not installable
                            Depends: ros-melodic-geometry-msgs but it is not installable
                            Depends: ros-melodic-nav-msgs but it is not installable
                            Depends: ros-melodic-rosbash but it is not installable
                            Depends: ros-melodic-roscpp but it is not installable
                            Depends: ros-melodic-roscpp-tutorials but it is not installable
                            Depends: ros-melodic-roslaunch but it is not installable
                            Depends: ros-melodic-rosmsg but it is not installable
                            Depends: ros-melodic-rospy-tutorials but it is not installable
                            Depends: ros-melodic-sensor-msgs but it is not installable
                            Depends: ros-melodic-std-msgs but it is not installable
                            Depends: ros-melodic-std-srvs but it is not installable
                            Depends: ros-melodic-stereo-msgs but it is not installable
                            Depends: ros-melodic-tf2-msgs but it is not installable
                            Depends: ros-melodic-trajectory-msgs but it is not installable
                            Depends: ros-melodic-visualization-msgs but it is not installable
E: Unable to correct problems, you have held broken packages.
```

jacobperron, 2020-10-28:
I can reproduce this issue. The problem occurs if you haven't set up your source.list for ROS 1 repositories. IMO, this shouldn't be a prerequisite for the tutorial and we should amend the install command to avoid installing `ros-<distro>-rosbag2-bag-v2-plugins`, which shouldn't be a necessary package for the tutorial.

jacobperron, 2020-10-28:
See https://github.com/ros2/ros2_documentation/pull/896 for proposed changes to the tutorial instructions.

hsyis, 2020-10-28:
@jacobperron I followed the changes in #896 and it works fine. Thank you :)

PR #896 (2020-10-28): Update rosbag2 tutorial to avoid installing ROS 1 as transitive dependency
Resolves #879

Previously, the glob install for ros-<distro>-rosbag2* was inadvertantly installing the ROS 1 plugin package
rosbag2_bag_v2_plugins. The install command may fail if the users sources.list is not set up for ROS 1.
commit 52ca3997 (2020-10-28): Update rosbag2 tutorial to avoid installing ROS 1 as transitive dependency (#896)

Resolves #879

Previously, the glob install for ros-<distro>-rosbag2* was inadvertantly installing the ROS 1 plugin package
rosbag2_bag_v2_plugins. The install command may fail if the users sources.list is not set up for ROS 1.
closed 2020-10-28 by jacobperron