ros2/ros2_documentation #880: Mention best pratice for passing rcutils_allocator_t around in the code style

christophebedard, 2020-09-29:
Originally brought up here: https://github.com/ros2/rcl_logging/pull/53#discussion_r496798867

The best pratice seems is apparently to pass `rcutils_allocator_t` around using pointers, even if they are passed by value in some places.

There is a style guide for language-sepcific things (https://index.ros.org/doc/ros2/Contributing/Code-Style-Language-Versions/) but is there a ROS-specific style guide/best pratices document? Maybe we can add a new section to the developer guide? https://index.ros.org/doc/ros2/Contributing/Developer-Guide/

I can open a PR, but if someone could comment on where I should add this, that would be great!

wjwwood, 2020-09-29:
I'd dispute that this is a preferred pattern in all cases. Passing by value is what I would do in most cases. We never came to a conclusion on this I think.

christophebedard, 2020-09-29:
oh, well, if there is no consensus then I can either close this or wait for a consensus.

christophebedard, 2020-09-30:
We'll go with this as the consensus: https://github.com/ros2/rcl_logging/pull/53#discussion_r497058492