ros2/ros2_documentation #881: Failed to load entry point 'pub': librclpy_common.so: cannot open shared object file: No such file or directory

ActivePeter, 2020-09-30:
run on nanopi neo4 (rk3399  arm64 ubuntu18)

The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'find': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'list': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'type': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'bw': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'delay': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'echo': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'find': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'hz': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'info': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'list': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'pub': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'type': librclpy_common.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-aarch64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
/opt/ros/eloquent/lib/demo_nodes_cpp/talker: error while loading shared libraries: libclass_loader.so: cannot open shared object file: No such file or directory
pi@NanoPi-NEO4:~$

ActivePeter, 2020-09-30:
The command is ros2 run demo_nodes_cpp talker

clalancette, 2020-09-30:
Can you please give us more information about your environment?  In particular, which ROS distribution you installed, how you installed it, and what your Python environment is like (are you using conda, venvs, etc)?

ActivePeter, 2020-09-30:
https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Install-Debians/  I followed this tutorial step by step. The python environment is pip.

ActivePeter, 2020-09-30:
I changed to use ros1 now and it is no problem

clalancette, 2020-09-30:
Can you run `python3 --version` on the system, and `pip3 freeze`?

ActivePeter, 2020-09-30:
Python 3.6.8

ActivePeter, 2020-09-30:
actionlib==1.12.1
angles==1.9.12
argcomplete==1.12.1
asn1crypto==0.24.0
attrs==17.4.0
beautifulsoup4==4.6.0
bondpy==1.8.5
catkin==0.7.28
catkin-pkg-modules==0.4.22
certifi==2018.1.18
chardet==3.0.4
cryptography==2.1.4
cupshelpers==1.0
cv-bridge==1.13.0
cycler==0.10.0
defer==1.0.6
diagnostic-analysis==1.9.7
diagnostic-common-diagnostics==1.9.7
diagnostic-updater==1.9.7
distro-info===0.18ubuntu0.18.04.1
docutils==0.14
dynamic-reconfigure==1.6.3
empy==3.3.2
flake8==3.5.0
gencpp==0.6.5
geneus==2.2.6
genlisp==0.4.16
genmsg==0.5.16
gennodejs==2.0.1
genpy==0.6.14
html5lib==0.999999999
httplib2==0.9.2
idna==2.6
ifcfg==0.18
importlib-metadata==2.0.0
interactive-markers==1.11.5
joint-state-publisher==1.12.15
kdl-parser-py==1.13.1
keyring==10.6.0
keyrings.alt==3.0
language-selector==0.1
lark-parser==0.7.2
laser-geometry==1.6.5
lxml==4.2.1
macaroonbakery==1.1.3
matplotlib==2.1.1
mccabe==0.6.1
message-filters==1.14.9
netifaces==0.10.4
numpy==1.13.3
olefile==0.45.1
pbr==5.2.0
Pillow==5.1.0
pluggy==0.6.0
protobuf==3.0.0
psutil==5.4.2
py==1.5.2
pycairo==1.16.2
pycodestyle==2.3.1
pycrypto==2.6.1
pycups==1.9.73
pydocstyle==2.0.0
pydot==1.2.3
pyflakes==1.6.0
Pygments==2.2.0
pygobject==3.26.1
pymacaroons==0.13.0
PyNaCl==1.1.2
pyparsing==2.2.0
pyRFC3339==1.0
pytest==3.3.2
python-apt==1.6.0
python-dateutil==2.6.1
python-qt-binding==0.4.3
pytz==2018.3
pyxdg==0.25
PyYAML==3.12
qt-dotgraph==0.4.2
qt-gui==0.4.2
qt-gui-cpp==0.4.2
qt-gui-py-common==0.4.2
requests==2.18.4
requests-unixsocket==0.1.5
resource-retriever==1.12.6
roman==2.0.0
rosbag==1.14.9
rosboost-cfg==1.14.9
rosclean==1.14.9
roscreate==1.14.9
rosdistro-modules==0.8.2
rosgraph==1.14.9
roslaunch==1.14.9
roslib==1.14.9
roslint==0.11.2
roslz4==1.14.9
rosmake==1.14.9
rosmaster==1.14.9
rosmsg==1.14.9
rosnode==1.14.9
rosparam==1.14.9
rospkg-modules==1.2.8
rospy==1.14.9
rosservice==1.14.9
rostest==1.14.9
rostopic==1.14.9
rosunit==1.14.9
roswtf==1.14.9
rqt-action==0.4.9
rqt-bag==0.4.14
rqt-bag-plugins==0.4.14
rqt-console==0.4.9
rqt-dep==0.4.9
rqt-graph==0.4.11
rqt-gui==0.5.2
rqt-gui-py==0.5.2
rqt-image-view==0.4.16
rqt-launch==0.4.8
rqt-logger-level==0.4.8
rqt-moveit==0.5.7
rqt-msg==0.4.8
rqt-nav-view==0.5.7
rqt-plot==0.4.9
rqt-pose-view==0.5.8
rqt-publisher==0.4.8
rqt-py-common==0.5.2
rqt-py-console==0.4.8
rqt-reconfigure==0.5.3
rqt-robot-dashboard==0.5.7
rqt-robot-monitor==0.5.13
rqt-robot-steering==0.5.10
rqt-runtime-monitor==0.5.7
rqt-rviz==0.6.0
rqt-service-caller==0.4.8
rqt-shell==0.4.9
rqt-srv==0.4.8
rqt-tf-tree==0.6.0
rqt-top==0.4.8
rqt-topic==0.4.11
rqt-web==0.4.8
rviz==1.13.13
SecretStorage==2.3.1
sensor-msgs==1.12.7
six==1.11.0
smach==2.0.1
smach-ros==2.0.1
smclib==1.8.5
snowballstemmer==1.2.1
ssh-import-id==5.7
stevedore==1.30.1
system-service==0.3
systemd-python==234
tf==1.12.1
tf-conversions==1.12.1
tf2-kdl==0.6.5
tf2-py==0.6.5
tf2-ros==0.6.5
topic-tools==1.14.9
unattended-upgrades==0.1
urdfdom-py==0.4.3
urllib3==1.22
virtualenv==16.5.0
virtualenv-clone==0.5.3
virtualenvwrapper==4.8.4
webencodings==0.5
wiringpi==2.44.4
xacro==1.13.8
zipp==3.2.0

clalancette, 2020-10-05:
So it seems like this is the main error:

```
Failed to load entry point 'list': librclpy_common.so: cannot open shared object file: No such file or directory
```

Do you have `ros-eloquent-rclpy` installed?  Can you run:

```
dpkg -L ros-eloquent-rclpy
```

clalancette, 2021-01-14:
I'm going to close this one out since I haven't heard back for a while.  But please feel free to keep commenting or reopen if you are still running into problems.

closed 2021-01-14 by clalancette

kumiusge, 2024-03-19:
In the pyCharm,    you will setup  the .. Run/Debug Configurations  ->Environment-> Environment variables:->
   +  -> Name: LD_LIBRARY_PATH,   Value: /opt/ros/foxy/lib