ros2/ros2_documentation #885: Missing quotes in "Installing ROS 2 on Windows" page

drewhamiltonasdf, 2020-10-08:
Minor issue, but the type of thing that can really trip up a new user like myself:

In "Install OPENSSL" section, the following

    setx -m OPENSSL_CONF C:\Program Files\OpenSSL-Win64\bin\openssl.cfg

should have quotes around the path, as in

    setx -m OPENSSL_CONF "C:\Program Files\OpenSSL-Win64\bin\openssl.cfg"

or it will throw an error: ERROR: Invalid syntax. Default option is not allowed more than '2' time(s).
Type "SETX /?" for usage. "

Many thanks to all the folks in the trenches tirelessly fielding these comments and making minor adjustments!

Best,
Drew Hamilton

clalancette, 2020-10-08:
Oh, good point.  Would you mind opening a pull request with that change?  Thanks.

clalancette, 2020-10-22:
This was fixed now, so closing.

closed 2020-10-22 by clalancette