ros2/ros2_documentation #892: wiki.ros.org does not show ROS2 distros

nnmm, 2020-10-21:
I hope this is the right repo to file this issue.

In all packages I've seen so far on wiki.ros.org, the list of distros at the top only includes ROS1 distros, even if that package is available for ROS2. That makes it hard to tell whether a package is available for ROS2, and where to find documentation. I was only able to find package documentation for ROS2 distros by googling the package name. Could the ROS2 docs be made easily accessible from there, or at least include a note where to look for ROS2 documentation?

clalancette, 2020-10-21:
That is by design.  For ROS 2, we are deprecating wiki.ros.org.

If you'd like to look up packages across all distributions, your best bet is to use https://index.ros.org.  That is an "index" of all available ROS packages across all distributions.

I'm going to close this out for now, but please feel free to keep commenting.

closed 2020-10-21 by clalancette

nnmm, 2020-10-21:
Thank you for your quick reply! In that case, I think pointing that out would be a great improvement to user-friendliness. It is very easy for a newcomer to miss that.

E.g. when googling for "tf2_ros", wiki.ros.org is one of the first results, index.ros.org is not even on the first page ("tf2_ros ros2" is a better search term of course, but still). The wiki's URL doesn't indicate that it's for ROS1 only, and nowhere on a package page or even the wiki.ros.org home page is it mentioned.

I'd propose something like this (crappy mockup): 

[image]

It could open a drop-down like the EOL distros checkbox does. If that seems like it would be worth it, I can make such a change if you point me to the code.