ros2/ros2_documentation #927: Beginner configuring tutorial: workspace instruction

rpcme, 2020-11-26:
The beginner tutorial

https://index.ros.org/doc/ros2/Tutorials/Configuring-ROS2-Environment/#configros2

Has lots of good information about setting the environment up, but nothing about /creating/ a workspace.

The prior tutorial (installation) does not include it.

The subsequent tutorial does directly into package creation, but without a workspace ... not much success.

sheila-suppiah, 2022-07-06:
@shalman-khan

shalman-khan, 2022-07-06:
The [Beginner: CLI tools tutorial](https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html ) doesn't need to have a prerequisite to include workspace creation tutorial. Also the subsequent tutorial is about package installation [not package creation] which directly installs packages in /opt/ros/$ROS_DISTRO. The package creation tutorial is in [Beginner:Client Libraries tutorial ](https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries.html) section, which does have a [workspace creation tutorial](https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html) before [package creation tutorial](https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.html).

clalancette, 2022-07-06:
Yes, I believe that we've more-or-less fixed this as part of the recent revamp of the tutorials.  I'm going to close this out, but feel free to reopen if you think that this should still be changed further.

closed 2022-07-06 by clalancette