ros2/ros2_documentation #935: Is creating an action only supported in C++?

dqii, 2020-12-09:
There is only documentation for creating an action in C++, but writing an action server and client is supported in Python. Does creating an action need to be done in C++?

clalancette, 2020-12-09:
There's a whole tutorial on how to do it in Python: https://index.ros.org/doc/ros2/Tutorials/Actions/Writing-a-Py-Action-Server-Client/

closed 2020-12-09 by clalancette

dqii, 2020-12-09:
I saw the tutorial for writing an action server and client. When I said "creating an action" I was referencing
https://index.ros.org/doc/ros2/Tutorials/Actions/Creating-an-Action/
I was confused because it used CMakeLists.txt, which didn't seem to show up in any of the other Python tutorials, so I was not sure if there was supposed to be a separate tutorial for "creating an action" in Python. But it seems like the CMakeLists.txt should be created in Python as well. Sorry for the confusion.