ros2/ros2_documentation #1066: [Windows] ROS2 development with Visual Studio

ItsMeTheBee, 2021-01-29:
## Feature request

#### Feature description
It would be great if you could provide a little guide on how to work with ROS 2 in Visual Studio. There are some questions like [this one](https://answers.ros.org/question/340277/ros2-windows-running-c-node-by-direct-dlllib-reference/) on answers.ros but I haven´t been able to get it to run yet.

Right now my setup is like this:

- C/C++ > Additional Include Directories = C:\dev\ros2_foxy\include
- Linker > Additional Library Directories = C:\dev\ros2_foxy\Lib
- Linker > Input > Additional Dependencies = all .lib files in C:\dev\ros2_foxy\Lib

But I still get a bunch of LNK2019 errors. I guess I´m still missing something but according to the linked question there are other problems as well even if there are no LNK errors.

I think a tutorial for Visual Studio would be great so more people can incorporate ROS into other projects where colcon is not an option.

ItsMeTheBee, 2021-02-02:
@clalancette 
Not absolutely sure what the problem is since I can`t debug into ROS 2 because there is no pdb. Do you happen to have pdb files at hand? It would make the whole isse a lot easier. I guess I can try to build ROS inside of Visual Studio instead of using the .lib and include files to create the pdbs myself but to be honest I never tried to create pdbs so I have no idea.

Apart from that it seems like I´m getting a `rclcpp.dll!rclcpp::exceptions::RCLInvalidROSArgsError::RCLInvalidROSArgsError()` exception. I build the example publisher with colcon to look at argv and it contained the path to the .exe of the node. So I build the equivalent myself:

```
 std::vector<std::string> arguments = {"D:\\UserData\\z003y2wv\\VisualStudio\\NX_CSharp\\x64\\Debug\\RosNoCli.exe"};
    std::vector<char*> argv;
    for (const auto& arg : arguments)
        argv.push_back((char*)arg.data());

    rclcpp::init(argv.size() - 1, argv.data());
```

But I still get the same exception. Maybe you can provide some insight here?

clalancette, 2021-02-02:
> Not absolutely sure what the problem is since I can`t debug into ROS 2 because there is no pdb.

I don't know what the pdb is, sorry.

> `rclcpp.dll!rclcpp::exceptions::RCLInvalidROSArgsError::RCLInvalidROSArgsError()` exception. I build the example publisher with colcon to look at argv and it contained the path to the .exe of the node. So I build the equivalent myself:

If you just want to test things out, you can pass a 0 and nullptr to `rclcpp::init`, like this:

```
rclcpp::init(0, nullptr);
```

ItsMeTheBee, 2021-02-02:
I tried `rclcpp::init(0, nullptr);` as well but that results on the same exception/ error.

ooeygui, 2022-04-25:
Instead of Visual Studio, would it be acceptable to write a tutorial here for using the Visual Studio Code ROS extension?

ItsMeTheBee, 2022-04-26:
I actually managed to solve the issue, I can't remember what fixed it for me but I created a repo with a sample Visual Studio solution and some sample code - I could provide a link if it's interesting for you but you can also find it through my profile. 

Oh and I couldn't use Visual Studio Code due to project requirements.

closed 2022-04-26 by ItsMeTheBee