ros2/ros2_documentation #1071: colcon_cd is referenced before colcon is installed

tykurtz, 2021-01-31:
For reference, I'm walking through the documentation blind for the first time.

From https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/#next-steps-after-installing I'm brought to https://index.ros.org/doc/ros2/Tutorials/ where I meet

> Please walk through the tutorials for the first time in order, they build off of each other and are not meant to be comprehensive documentation

which I agree to follow. This adds confusion when I encounter https://index.ros.org/doc/ros2/Tutorials/Configuring-ROS2-Environment/#add-colcon-cd-to-your-shell-startup-script in the first tutorial as `colcon` has not been referenced much less installed up until this point.

There's a couple ways this could be fixed.

* Move the colcon_cd configuration to the colcon tutorial
* Add a link to the install colcon page on the install ROS2 pages
* Add a link to the install colcon page from the configuring ROS2 environment page. This seems less than ideal as now we're jumping tutorials out of order

pxalcantara, 2021-07-11:
Is there a preferred approach among the listed above to solve the `colcon_cd` issue @clalancette?

pxalcantara, 2021-07-18:
> Is there a preferred approach among the listed above to solve the `colcon_cd` issue @clalancette?

friendly ping @clalancette

sheila-suppiah, 2022-07-06:
@shalman-khan

shalman-khan, 2022-07-06:
The latest documentation for colcon_cd setup  is placed at the right location, after colcon installation instructions. Please refer to this link (https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html). The colcon install and colcon_cd setup documentation is also placed at the correct location, before the package creation and build tutorials. Refer Link: (https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries.html).

clalancette, 2022-07-06:
Yes, agreed, this got fix with some of our recent refactoring.  Closing this out, but feel free to reopen if you feel this is still a problem.

closed 2022-07-06 by clalancette