ros2/ros2_documentation #1093: Add a common header to all pages in the documentation

tfoote, 2021-02-09:
On other sites we've adopted this shared header:

```
<div align="center">
      <br>
      <table id="topnav-table">
        <tr>
          <td valign="middle">
              ROS Resources: 
            <a href="http://wiki.ros.org/">Documentation</a>
            |
            <a href="http://wiki.ros.org/Support">Support</a>
            |
            <a href="http://discourse.ros.org/">Discussion Forum</a>
            |
            <a href="http://status.ros.org/">Service Status</a>
            |
            <a href="http://answers.ros.org/">Q&A answers.ros.org</a>

          </td>

      </tr>
    </table> <!-- topnav-table -->
</div>
```

This sort of thing would be valuable to have as we deploy the standalone site. We're working on a more complete one with better theming, but for now this should get people a consistent navigation experience.

clalancette, 2021-02-10:
In general, I agree with the banner idea.  However, I don't think we should put the links to http://wiki.ros.org (that would be the Documentation and Support links).  We are explicitly moving away from those in ROS 2, so adding additional links to it seems counterproductive.

clalancette, 2021-02-10:
This was done in #1096, so closing.

closed 2021-02-10 by clalancette