ros2/ros2_documentation #1110: Create guides for ROS2 launch and ROS2 parameters

audrow, 2021-02-18:
In my understanding of where we are going with the ROS 2 documentation, guides are different than tutorials in that they are not necessarily examples for readers to follow along, but rather are meant to be concise references for doing specific things in ROS 2. 

The following guides could be written for the ROS 2 documentation. Each checkbox is intended to be one guide, with sublists providing more detail about what content could be included in the guide.

## ROS2 Launch
- [ ] Using parameters in launch files
  - Parameters from config file
  - Define parameters in a launch file
  - Pushing into a namespace
- [ ] Launching other launch files
  - Simple include
  - Remapping
  - Namespaces
- [ ] Comparing launch files written in YAML, XML, and Python
  - Show all three examples
    - Node
    - Namespace
    - Set parameter
    - Args
    - Substitution
- [ ] Using substitution in launch files
  - Text
  - Environment variable
  - Executable

## ROS2 Parameters
- [ ] Parameter files
  - Wildcards
  - Namespaces

audrow, 2021-02-24:
Additional thoughts:
* Cover Python, XML, and YAML for all ROS2 launch guides
* Show parameter override precedence in Node action

clalancette, 2021-02-24:
>     * Cover Python, XML, and YAML for all ROS2 launch guides

Ivan made a good start with that in this PR: https://github.com/ros2/ros2_documentation/pull/1109 .  Probably we can just work on and expand that one.

audrow, 2021-05-05:
It would also be good to have DDS troubleshooting across hosts.

audrow, 2021-06-08:
It would be great to add how to use launch with a parameter file. I think a good spot for this is here: https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Parameters/Understanding-ROS2-Parameters.rst#7-load-parameter-file.

audrow, 2022-02-16:
All the items have been added!

closed 2022-02-16 by audrow