ros2/ros2_documentation #1131: Install ROS2 on macOS fails with pygraphviz

stephenstarkie, 2021-02-23:
I tried to install according to the instructions here: https://index.ros.org/doc/ros2/Installation/Foxy/macOS-Install-Binary/

Installation fails on the line:

python3 -m pip install pygraphviz pydot

with;

Collecting pygraphviz
  Using cached pygraphviz-1.7.zip (118 kB)
Building wheels for collected packages: pygraphviz
  Building wheel for pygraphviz (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/local/opt/python@3.8/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/7x/9ydhg6pn74s30xzlsltt5h1w0000gp/T/pip-install-j0l_0s3c/pygraphviz_ac93efbdfc764377ae19b8ff1b8c3d95/setup.py'"'"'; __file__='"'"'/private/var/folders/7x/9ydhg6pn74s30xzlsltt5h1w0000gp/T/pip-install-j0l_0s3c/pygraphviz_ac93efbdfc764377ae19b8ff1b8c3d95/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/7x/9ydhg6pn74s30xzlsltt5h1w0000gp/T/pip-wheel-gsy173sm
       cwd: /private/var/folders/7x/9ydhg6pn74s30xzlsltt5h1w0000gp/T/pip-install-j0l_0s3c/pygraphviz_ac93efbdfc764377ae19b8ff1b8c3d95/
  Complete output (311 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-11-x86_64-3.8
  creating build/lib.macosx-11-x86_64-3.8/pygraphviz
  copying pygraphviz/scraper.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz
  copying pygraphviz/graphviz.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz
  copying pygraphviz/__init__.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz
  copying pygraphviz/agraph.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz
  copying pygraphviz/testing.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz
  creating build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_unicode.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_scraper.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_readwrite.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_string.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/__init__.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_html.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_node_attributes.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_drawing.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_subgraph.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_close.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_edge_attributes.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_clear.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_layout.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_attribute_defaults.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  copying pygraphviz/tests/test_graph.py -> build/lib.macosx-11-x86_64-3.8/pygraphviz/tests
  running egg_info
  writing pygraphviz.egg-info/PKG-INFO
  writing dependency_links to pygraphviz.egg-info/dependency_links.txt
  writing top-level names to pygraphviz.egg-info/top_level.txt
  reading manifest file 'pygraphviz.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.png' under directory 'doc'
  warning: no files found matching '*.txt' under directory 'doc'
  warning: no files found matching '*.css' under directory 'doc'
  warning: no previously-included files matching '*~' found anywhere in distribution
  warning: no previously-included files matching '*.pyc' found anywhere in distribution
  warning: no previously-included files matching '.svn' found anywhere in distribution
  no previously-included directories found matching 'doc/build'
  writing manifest file 'pygraphviz.egg-info/SOURCES.txt'
  copying pygraphviz/graphviz.i -> build/lib.macosx-11-x86_64-3.8/pygraphviz
  copying pygraphviz/graphviz_wrap.c -> build/lib.macosx-11-x86_64-3.8/pygraphviz
  running build_ext
  building 'pygraphviz._graphviz' extension
  creating build/temp.macosx-11-x86_64-3.8
  creating build/temp.macosx-11-x86_64-3.8/pygraphviz
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/opt/tcl-tk/include -I/usr/local/Cellar/python@3.8/3.8.8/Frameworks/Python.framework/Versions/3.8/include/python3.8 -c pygraphviz/graphviz_wrap.c -o build/temp.macosx-11-x86_64-3.8/pygraphviz/graphviz_wrap.o
  In file included from pygraphviz/graphviz_wrap.c:154:
  In file included from /usr/local/Cellar/python@3.8/3.8.8/Frameworks/Python.framework/Versions/3.8/include/python3.8/Python.h:34:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdlib.h:66:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/resource.h:72:
  In file included from /usr/local/include/stdint.h:59:
  In file included from /usr/local/include/stdint.h:59:
...lots of the same line
In file included from /usr/local/include/stdint.h:59:
  In file included from /usr/local/include/stdint.h:59:
  /usr/local/include/stdint.h:2:10: error: #include nested too deeply

stephenstarkie, 2021-02-23:
macos 11.2.1

clalancette, 2021-02-23:
We don't officially support Big Sur with any of our platforms, so it is not entirely surprising.  That being said, this looks like a problem in pygraphviz, nothing specific to ROS 2 here.  So I'm going to close this out; I'll suggest that you open an issue against https://github.com/pygraphviz/pygraphviz instead.

closed 2021-02-23 by clalancette