ros2/ros2_documentation #1160: Have a nicer looking front-page at https://docs.ros.org

clalancette, 2021-03-02:
Right now if you visit that site, you are met with a directory listing.  In a [discussion on Discourse](https://discourse.ros.org/t/docs-ros2-org-is-being-retired/18917/5), there were some opinions that we should have a nicer landing page, and some that we should keep the ability to get the directory listing.  Here are some things we might consider:

1. Just having a landing page that has links into the directory hierarchy.  This is nicer for users first arriving, and is pretty common.  It has the downside that it is no longer easy/possible to look at the raw directory listing.
2. Leave the current directory listing as-is.  This has the obvious downside that it is confusing for users who first arrive.  It has the benefit that more advanced users can poke around in the structure.
3. Prettify the current directory listing.  There are ways to make that directory listing a lot more pretty.  This has some of the benefits of 1 and 2 above.

There are probably other things we can do here.

christophebedard, 2021-03-02:
Let me know if I should open a separate issue (or if it already exists), but it is kind of related: would it be possible to have a "permanent dynamic redirect" to the latest distro? For example, https://docs.ros.org/en/latest/ which would redirect to https://docs.ros.org/en/foxy/ now and to https://docs.ros.org/en/galactic/ when it gets released. /latest exists (so do /latest-available and /latest-lts), but it contains other stuff.

Having a direct link (without needing to interact with a docs.ros.org front page) would be great.

clalancette, 2021-03-02:
> Let me know if I should open a separate issue (or if it already exists), but it is kind of related: would it be possible to have a "permanent dynamic redirect" to the latest distro? For example, [...]

Yeah, it's a good idea.  However, since https://docs.ros.org is used to serve content for both ROS 1 and ROS 2, there is a conflict; the 'latest' symlink already points to the latest ROS 1 distribution, and we don't want to break that link.

Probably the way out of this is to add a 'latest-ros1' and 'latest-ros2' link there, and use that.

christophebedard, 2021-03-02:
> Probably the way out of this is to add a 'latest-ros1' and 'latest-ros2' link there, and use that.

yeah that sounds good!

tfoote, 2021-03-03:
RE: latest

Adding multiple paths to the same content increases the cognative overload and now we could end up with potentially 3 different url structures to the same release that people would need to be aware of.

I actually think that we might not want to add all these several redirects. Now that we have rolling, it is effectively "latest". If you want to give latest a unique meaning. you would have to then be the most recent one that's not rolling. 

But then I look at what providing a link to "latest" means. It means that I want content that's relevant now and is expected to be relevant in the future releases. But in that case it should be relevant in rolling, because if it's not relevant in rolling than a link to "latest" will go invalid as soon the next rosdistro comes out following our above definition that your link is not compatible with rolling. In which case you should not point to "latest" but should point to the specific distro for which your pointer is relevant. 

We also are doing much better version visibility and cross referencing so you can now click from one version of the docs to the other quickly. We plan to do the same for the per package documentation that's still in progress. 

All that put together I'd suggest that we keep things simpler and not start adding aliases and redirects to this effect.

tfoote, 2021-03-03:
For the landing pages and front page we might be able to leverage custom DirectoryListings: https://cwiki.apache.org/confluence/display/HTTPD/DirectoryListings to get both the browsing as well as a standardized header.

christophebedard, 2021-03-03:
> Adding multiple paths to the same content increases the cognative overload and now we could end up with potentially 3 different url structures to the same release that people would need to be aware of.

That's fair.

I was just thinking that plenty of projects/websites want to link to the latest released distro, and for most people, latest isn't really Rolling. So maybe I mean "latest LTS." 

In any case, it was really just a suggestion.

clalancette, 2021-03-17:
All right, we just landed something that makes https://docs.ros.org have a landing page.  It isn't super pretty, but at least it isn't a directory listing anymore.  So I'm going to close this out for now.

closed 2021-03-17 by clalancette

gavanderhoorn, 2021-03-17:
Accessing the index doesn't seem possible any more @clalancette.

That's by-design?

clalancette, 2021-03-17:
> Edit: oh, it seems [docs.ros.org/en](http://docs.ros.org/en) does still show the index.

Yes, exactly.  Hopefully that suffices for your use.