ros2/ros2_documentation #1170: Corporate logo should be size aligned to have nicer looking?

fujitatomoya, 2021-03-03:
I'd like to open discussion for this. as a visual identification, i think it would be nice to have the same (or almost) view size for corporate logo on [Governance](https://docs.ros.org/en/foxy/Governance.html). at this moment, some are big, some are small.

fujitatomoya, 2021-03-03:
@clalancette friendly ping.

clalancette, 2021-03-03:
Yes, they are all meant to be (approximately) the same size.  Somehow it looks like the new Read-the-Docs theme is rendering them differently than before.  So this is something that we should fix.  Any contribution here is welcome.

fujitatomoya, 2021-03-03:
i am not sure how to fix this, i'll ask around 😅

tfoote, 2021-03-03:
For someone looking into this, it looks like the the 35px height element is being overridden in the new theme

[image]

PR #1218 (2021-03-11): Fix the TSC table so the logos are all approximately the same size.
I couldn't figure out how to do this with the RTD theme, so
I ended up making the whole table in raw HTML/CSS.  All of the
logos are now approximately the same size.

This should fix #1170 .  FYI @fujitatomoya
closed 2021-03-11 by clalancette