ros2/ros2_documentation #1176: Tutorials: Write a QoS concept, guide, and tutorial

clalancette, 2021-03-03:
Dealing with Quality of Service settings is very important to being successful with ROS 2.  We should write a series of 3 articles dealing with QoS: a concept article, a guide, and a tutorial.

Here are some notes on these articles:

# Concepts

Add “why” you might want to change those settings, situations, benefits, etc. section under “QoS policies”

# Guide

* How to work with the rclcpp::QoS and rclpy.QoS
* (for each) Setting
    * How to set (code snippet) for each possible value
    * CLI implications

c++ code

```
rclcpp::SubscriptionOptions so;
so.event_callbacks.deadline_callback = [](...) { …; };
sub = create_sub(..., rclcpp::QoS(10).deadline(1s), so);
```

# Tutorial
* New Tutorial 1
    * extending the pub/sub tutorial to demonstrate a few QoS settings, then use turtlesim to show off one or more QoS in action
    * goal:
        * introduce the idea of QoS
        * introduce the idea that there are lots of QoS settings that can be changed and how that can change their project
        * how to change QoS using C++/Python (forked tutorial)
        * show CLI tools can have QoS settings changed too
    * which QoS should be used:
        * history and depth (no)
            * because it’s hard to demo with just CLI and turtlesim
        * reliability (no)
            * difficult to demo without modifying the network
        * durability (yes)
            * easy to show based on when you start the nodes
        * deadline (yes)
            * it’s good for turtlesim (can be used to stop the turtle)
        * lifespan (maybe)
            * you could combine with durability to show, but a bit complicated
        * liveliness (yes)
            * you can just shutdown one and the other will notice
        * Compatibility notice
            * Purposefully use mismatched qos settings to show the message that comes up

FelipeGdM, 2023-12-07:
A lot of time has passed since this issue was created, I was wondering if the context has changed since then

How does this issue relates to the existing QoS related content in the docs? Like [Quality of Service settings](https://docs.ros.org/en/iron/Concepts/Intermediate/About-Quality-of-Service-Settings.html) page

I believe the concept article may be covered, but the guide and the tutorial are still needed

clalancette, 2023-12-07:
> I believe the concept article may be covered, but the guide and the tutorial are still needed

Yeah, I think that is the case.  Given my evolving understanding of the documentation, I think that we would need some Tutorials specifically pointing out how each of the QoS settings works, and how they interact with each other.  If we wrote that series of tutorials, then I think we could close this issue out.