ros2/ros2_documentation #1192: Minor update request for 'Building realtime Linux for ROS 2' tutorial

JaehyunShim, 2021-03-09:
Hi, 

I followed the instruction in this [link](https://docs.ros.org/en/foxy/Tutorials/Building-Realtime-rt_preempt-kernel-for-ROS-2.html) and needed to do the following two things in addition to the tutorial.

1. sudo apt-get update after ticking the ‘Source code’ box in Software & Updates. 
2. Turned off Secure Boot in the BIOS when rebooting.

clalancette, 2021-03-09:
Please feel free to open a pull request with any updates you think are necessary.

I will say, however, that I don't think we should suggest that people turn off Secure Boot.  There are tutorials online on how to make a custom kernel work with Secure Boot.

JaehyunShim, 2021-03-10:
> I will say, however, that I don't think we should suggest that people turn off Secure Boot. There are tutorials online on how to make a custom kernel work with Secure Boot.

let me look it up first. Thanks