ros2/ros2_documentation #1199: Robot specific tutorials hosting

v-lopez, 2021-03-09:
In ROS1 the wiki.ros.org allowed hosting tutorials for public packages of various robot manufacturers. 

In our case we tutorials for many of our robots like http://wiki.ros.org/Robots/TIAGo/Tutorials/ which also contain examples of ROS packages such as ros_control, moveit and navigation applied to our robots. All executable with open source repositories.

What's the plan for ROS2? I've seen that this repository holds mostly generic ROS2 information, so these kind of tutorials would not belong here. 

Should we follow an approach similar to https://navigation.ros.org/? Would our content be accepted in the .ros.org domain, for instance as pal-robotics.ros.org or something similar?

If not, how could new users find such information?
We obviously benefit from reaching a wider user base, but we also believe that developers benefit from having tutorials with a wider range of robots with different capabilities such as mobile manipulators or biped robots.

tfoote, 2021-03-17:
For more specific documentation our recommendation is to embed it into the repository in rst or markdown files. We are in the process of enabling per package documenation builds in ROS 2 just like we've had in ROS 1. But our emphasis will be to keep the documentation closer to the code in the repository so the maintainers can keep it in sync better, and it will be versioned etc. That content will then all be put on to docs.ros.org by the documentation jobs in the package specific areas. And we plan to be able to add pointers to the packages homepages on index.ros.org

We can do the subdomains but I don't think that we want a very large proliferation of then for every robot or company, they're focused on larger community efforts. I would suggest making a package tiago_tutorials and putting all the content into that. Then you can have reference code and the tutorial contents all in one convenient place. It will be able to be rendered into docs.ros.org/en/DISTRO/p/tiago_tutorials publically, but you could also set it up to render to be generated to tutorials.pal-robotics.com or elsewhere with the same doc generation mechanism we'll be using on the farm through a github action etc.

v-lopez, 2021-03-17:
Perfect, thank you for the detailed explanation. We'll do that.

closed 2021-03-17 by v-lopez