ros2/ros2_documentation #1201: Foxy Latest Development Setup installation links broken

ammaar8, 2021-03-10:
The installation guide links for building from source at https://docs.ros.org/en/foxy/Installation/Latest-Development-Setup.html are broken.
`The requested URL /en/foxy/rolling/Installation/Linux-Development-Setup.html was not found on this server.`
If possible I'd like to try and fix this as it looks like it might be an easy-fix.

clalancette, 2021-03-10:
Yep, they definitely look broken.

And while we could definitely fix that up, I think it may make more sense to replace the entire contents of that page with a single line that says something like:

"Click here to get to the instructions for installing the latest development version of ROS 2".

With a link to https://docs.ros.org/en/rolling/Installation/Latest-Development-Setup.html

ammaar8, 2021-03-10:
Okay I will do that.

clalancette, 2021-03-10:
I'm honestly not sure.  An internal link is generally preferred, but there may be some weirdness because linking between different branches is something we don't do very much.

ammaar8, 2021-03-10:
I was able to get internal links and have created a PR. Please check. #1204 

Even though this is working won't we have extra pages which are not being used in all branches (other than the latest one) after backporting?

clalancette, 2021-03-10:
> Even though this is working won't we have extra pages which are not being used in all branches (other than the latest one) after backporting?

The only page that I see in that category would be the `Prerelease-Testing` page.  And we should indeed remove that one from the `foxy` branch (I'll make a comment on the PR).  Are there any others that you see?

ammaar8, 2021-03-10:
No none other than that. Sorry I misunderstood the pages a bit.

clalancette, 2021-03-10:
Yes please!

PR #1204 (2021-03-10): Add link in Latest-Development-Setup which points to rolling page
Fixes #1201. Replaces page content with single link which points to the `Latest-Development-Setup` page for  `rolling`.
closed 2021-03-10 by ammaar8