ros2/ros2_documentation #1277: Issue with the urdf_parser

kappa95, 2021-03-27:
I am upgrading to ROS2 but I have some issue with URDFs, I tried to do the ros2 tutorial: urdf_tutorial and I have issues regarding the `urdf_parser` in which is showing these errors when I'm opening rviz2:
```bash
[INFO] [1616839261.626448863] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1616839261.626652115] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1616839261.658635699] [rviz2]: Stereo is NOT SUPPORTED
Parsing robot urdf xml string.
Error:   length [.5] is not a valid float
         at line 183 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   Could not parse visual element for Link [axis]
         at line 495 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   length [.6] is not a valid float
         at line 183 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   Could not parse visual element for Link [body]
         at line 495 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   radius [.4] is not a valid float: stod
         at line 126 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   Could not parse visual element for Link [head]
         at line 495 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   length [.2] is not a valid float
         at line 183 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   Could not parse visual element for Link [rod]
         at line 495 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/link.cpp
Error:   lower value (-.5) is not a float: stod
         at line 130 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/joint.cpp
Error:   Could not parse limit element for joint [tilt]
         at line 565 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/joint.cpp
Error:   joint xml is not initialized correctly
         at line 206 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/model.cpp

```

Also in RViz I get the model moving and I'm seeing only the TFs moving:
[image]

I am using ROS2 Foxy on Ubuntu 20.04 LTS 64bit

clalancette, 2021-03-27:
For what it is worth, I just tried running through that tutorial on Foxy, and it seemed to work for me.  That tutorial could use some updating, which I've done in #1278, but there was nothing fundamentally wrong with it.

I'll suggest looking more closely at the URDF file you downloaded and make sure it matches the one from the tutorial.

kappa95, 2021-03-27:
I thought the same, infact I cloned the repo of the tutorial directly but I got the same

kappa95, 2021-03-27:
> For what it is worth, I just tried running through that tutorial on Foxy, and it seemed to work for me. That tutorial could use some updating, which I've done in #1278, but there was nothing [...]

I checked again the URDF and tried also to rename it into r2d2.urdf. But it seems correct and it is the same cloned from github. Could be a bug of `urdfdom/urdf_parser`?

clalancette, 2021-03-27:
> I checked again the URDF and tried also to rename it into r2d2.urdf. But it seems correct and it is the same cloned from github. Could be a bug of `urdfdom/urdf_parser`?

It doesn't seem like it, since it works fine for me here.  Can you give a lot more details about exactly what you have installed, and the exact commands you ran to get to this point?

kappa95, 2021-03-28:
> It doesn't seem like it, since it works fine for me here. Can you give a lot more details about exactly what you have installed, and the exact commands you ran to get to this point?

First of all I had uninstalled my old version of ROS Kinetic with `sudo apt remove ros-*` and I have upgraded the Ubuntu distro from 16.04 to 20.04. Later on I have installed ROS2 Foxy from [debian package](https://docs.ros.org/en/foxy/Installation/Linux-Install-Debians.html). (The only part that I didn't is the export of locale because I have every fields with: `it_IT.UTF-8
`)

Regarding the package itself: I followed exactly the guide:
* create a new workspace folder with its `src` folder
* sourced ROS with: `source /opt/ros/foxy/setup.bash`
* inside `src` I have created the python package with `colcon`: `ros2 pkg create urdf_tutorial --build-type ament_python --dependencies rclpy`
* Created the urdf folder and putted inside the two files: the urdf and Rviz config
* Written the two python files as the guide
* using rosdep for checking and installing dependencies: `rosdep install -i --from-path src --rosdistro $ROS_DISTRO -y`
* build with colcon with the same command in the guide.

The tree of folders:
```bash
├── build
│   ├── COLCON_IGNORE
│   └── urdf_tutorial
│       ├── colcon_build.rc
│       ├── colcon_command_prefix_setup_py.sh
│       ├── colcon_command_prefix_setup_py.sh.env
│       ├── launch
│       │   ├── demo.launch.py -> /home/kappa95/dev_ws/src/urdf_tutorial/launch/demo.launch.py
│       │   └── rviz.launch.py -> /home/kappa95/dev_ws/src/urdf_tutorial/launch/rviz.launch.py
│       ├── package.xml -> /home/kappa95/dev_ws/src/urdf_tutorial/package.xml
│       ├── resource
│       │   └── urdf_tutorial -> /home/kappa95/dev_ws/src/urdf_tutorial/resource/urdf_tutorial
│       ├── setup.cfg -> /home/kappa95/dev_ws/src/urdf_tutorial/setup.cfg
│       ├── setup.py -> /home/kappa95/dev_ws/src/urdf_tutorial/setup.py
│       ├── share
│       │   └── urdf_tutorial
│       │       └── hook
│       │           ├── pythonpath_develop.dsv
│       │           ├── pythonpath_develop.ps1
│       │           └── pythonpath_develop.sh
│       ├── urdf
│       │   ├── r2d2.rviz -> /home/kappa95/dev_ws/src/urdf_tutorial/urdf/r2d2.rviz
│       │   └── r2d2.urdf.xml -> /home/kappa95/dev_ws/src/urdf_tutorial/urdf/r2d2.urdf.xml
│       ├── urdf_tutorial -> /home/kappa95/dev_ws/src/urdf_tutorial/urdf_tutorial
│       └── urdf_tutorial.egg-info
│           ├── dependency_links.txt
│           ├── entry_points.txt
│           ├── PKG-INFO
│           ├── requires.txt
│           ├── SOURCES.txt
│           ├── top_level.txt
│           └── zip-safe
├── install
│   ├── COLCON_IGNORE
│   ├── local_setup.bash
│   ├── local_setup.ps1
│   ├── local_setup.sh
│   ├── _local_setup_util_ps1.py
│   ├── _local_setup_util_sh.py
│   ├── local_setup.zsh
│   ├── setup.bash
│   ├── setup.ps1
│   ├── setup.sh
│   ├── setup.zsh
│   └── urdf_tutorial
│       ├── lib
│       │   ├── python3.8
│       │   │   └── site-packages
│       │   │       ├── __pycache__
│       │   │       │   └── site.cpython-38.pyc
│       │   │       ├── site.py
│       │   │       └── urdf-tutorial.egg-link
│       │   └── urdf_tutorial
│       │       └── state_publisher
│       └── share
│           ├── ament_index
│           │   └── resource_index
│           │       └── packages
│           │           └── urdf_tutorial
│           ├── colcon-core
│           │   └── packages
│           │       └── urdf_tutorial
│           └── urdf_tutorial
│               ├── demo.launch.py
│               ├── hook
│               │   ├── ament_prefix_path.dsv
│               │   ├── ament_prefix_path.ps1
│               │   ├── ament_prefix_path.sh
│               │   ├── pythonpath.dsv
│               │   ├── pythonpath.ps1
│               │   └── pythonpath.sh
│               ├── package.bash
│               ├── package.dsv
│               ├── package.ps1
│               ├── package.sh
│               ├── package.xml
│               ├── package.zsh
│               ├── __pycache__
│               │   └── demo.launch.cpython-38.pyc
│               ├── r2d2.rviz
│               ├── r2d2.urdf.xml
│               └── rviz.launch.py
├── log
│   ├── build_2021-03-28_10-59-47
│   │   ├── events.log
│   │   ├── logger_all.log
│   │   └── urdf_tutorial
│   │       ├── command.log
│   │       ├── stderr.log
│   │       ├── stdout.log
│   │       ├── stdout_stderr.log
│   │       └── streams.log
│   ├── COLCON_IGNORE
│   ├── latest -> latest_build
│   └── latest_build -> build_2021-03-28_10-59-47
└── src
    └── urdf_tutorial
        ├── demo
        │   └── r2d2_rviz_demo.gif
        ├── launch
        │   ├── demo.launch.py
        │   └── rviz.launch.py
        ├── package.xml
        ├── README.md
        ├── resource
        │   └── urdf_tutorial
        ├── setup.cfg
        ├── setup.py
        ├── test
        │   ├── test_copyright.py
        │   ├── test_flake8.py
        │   └── test_pep257.py
        ├── urdf
        │   ├── r2d2.rviz
        │   └── r2d2.urdf.xml
        └── urdf_tutorial
            ├── __init__.py
            ├── __pycache__
            │   ├── __init__.cpython-38.pyc
            │   └── state_publisher.cpython-38.pyc
            └── state_publisher.py

```

kappa95, 2021-03-29:
I think to have find the issue, but in order to be sure to have everything cleaner I did a clean install of OS and with ROS Dashing. It seems that if you have others `locale` different from `en_US` the parsing doesn't work. Also in another package of inverse pendulum I was seeing only the box but not the rest of the pendulum. Later changing the `locale` in `en_US-UTF-8` and rebooting, everything goes fine.

closed 2021-03-29 by kappa95