ros2/ros2_documentation #1441: ros2 testing examples

simutisernestas, 2021-04-27:
Are there any plans on adding node testing examples?

simutisernestas, 2021-04-27:
Thanks! If someone would be willing to give some guidance I could help with documenting it.

Achllle, 2024-09-05:
Also surprised to find there's no documentation regarding integration testing in ROS 2. Wanted to jot down my findings from looking into this to maybe save someone else some time. 

ROS 1 has [rostest](https://wiki.ros.org/rostest), which seems to be ported to ROS 2 with [ros_testing](https://github.com/ros2/ros_testing), but there's also [launch_testing](https://github.com/ros2/launch/tree/master/launch_testing) which in the examples seems to focus on externally monitoring a particular node and checking its std output, rather than actually running `TestCase`s or `gtest`s within a script. I can't find more than a few words of documentation for `ros_testing`.

Two relevant discourse.ros.org discussions:
* https://discourse.ros.org/t/test-framework-in-ros2/2667
* https://discourse.ros.org/t/tutorials-for-unit-and-integration-testing-of-ros-scripts-and-packages/20892

From which I took away:

1. _You can do whatever works for you_. It's good that there are options for those who want to develop custom setups or have preferences, but I'd guess that most people are just looking for **a** recommended solution with examples they can start working with. Whatever is easiest, best documented, most supported.
2. Autoware seems to have developed [their own solution](https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/integration-testing.html) for smoke testing and integration testing. I personally would prefer to use an official ROS 2 recommended approach.