ros2/ros2_documentation #1465: Missing packages for Galactic Long-Form changelog

clalancette, 2021-05-05:
I'm currently putting together a long-form changelog for Galactic.  I have a script that walks through a source directory, looking at all of the CHANGELOG.rst files.  While doing that, it prints out a list of packages that don't have CHANGELOG.rst files.  For each of the packages, I think it behooves us to 1) start making CHANGELOG entries for it in the future, and 2) doing some manual work now to make sure the changes for Galactic for these packages are reflected in the long-form changelog.

Packages without a changelog:
* [ ] ament_clang_format
* [ ] ament_clang_tidy
* [ ] ament_cmake
* [ ] ament_cmake_auto
* [ ] ament_cmake_clang_format
* [ ] ament_cmake_clang_tidy
* [ ] ament_cmake_copyright
* [ ] ament_cmake_core
* [ ] ament_cmake_cppcheck
* [ ] ament_cmake_cpplint
* [ ] ament_cmake_export_definitions
* [ ] ament_cmake_export_dependencies
* [ ] ament_cmake_export_include_directories
* [ ] ament_cmake_export_interfaces
* [ ] ament_cmake_export_libraries
* [ ] ament_cmake_export_link_flags
* [ ] ament_cmake_export_targets
* [ ] ament_cmake_flake8
* [ ] ament_cmake_gmock
* [ ] ament_cmake_google_benchmark
* [ ] ament_cmake_gtest
* [ ] ament_cmake_include_directories
* [ ] ament_cmake_libraries
* [ ] ament_cmake_lint_cmake
* [ ] ament_cmake_mypy
* [ ] ament_cmake_nose
* [ ] ament_cmake_pclint
* [ ] ament_cmake_pep257
* [ ] ament_cmake_pycodestyle
* [ ] ament_cmake_pyflakes
* [ ] ament_cmake_pytest
* [ ] ament_cmake_python
* [ ] ament_cmake_ros
* [ ] ament_cmake_target_dependencies
* [ ] ament_cmake_test
* [ ] ament_cmake_uncrustify
* [ ] ament_cmake_version
* [ ] ament_cmake_xmllint
* [ ] ament_copyright
* [ ] ament_cppcheck
* [ ] ament_cpplint
* [ ] ament_flake8
* [ ] ament_index_cpp
* [ ] ament_index_python
* [ ] ament_lint
* [ ] ament_lint_auto
* [ ] ament_lint_cmake
* [ ] ament_lint_common
* [ ] ament_mypy
* [ ] ament_pclint
* [ ] ament_pep257
* [ ] ament_pycodestyle
* [ ] ament_pyflakes
* [ ] ament_uncrustify
* [ ] ament_xmllint
* [ ] console_bridge_vendor
* [ ] domain_coordinator
* [ ] example_interfaces
* [ ] fastrtps
* [ ] foonathan_memory_vendor
* [ ] gmock_vendor
* [ ] google_benchmark_vendor
* [ ] gtest_vendor
* [ ] iceoryx_binding_c
* [ ] iceoryx_posh
* [ ] iceoryx_utils
* [ ] image_transport_tutorial
* [ ] orocos_kdl
* [ ] python_cmake_module
* [ ] ros2action
* [ ] ros2cli
* [ ] ros2cli_test_interfaces
* [ ] ros2component
* [ ] ros2doctor
* [ ] ros2interface
* [ ] ros2lifecycle
* [ ] ros2lifecycle_test_fixtures
* [ ] ros2multicast
* [ ] ros2node
* [ ] ros2param
* [ ] ros2pkg
* [ ] ros2run
* [ ] ros2service
* [ ] ros2topic
* [ ] ros_environment
* [ ] rosidl_generator_dds_idl
* [ ] test_cli
* [ ] test_cli_remapping
* [ ] test_communication
* [ ] test_quality_of_service
* [ ] test_rclcpp
* [ ] test_security
* [ ] tinyxml2_vendor
* [ ] tinyxml_vendor
* [ ] uncrustify_vendor
* [ ] urdfdom

ivanpauno, 2021-05-05:
> doing some manual work now to make sure the changes for Galactic for these packages are reflected in the long-form changelog.

I'm not sure what's the exact output we're expecting from the manual work.
I will experiment with `ament_clang_format` locally to see what I can do.

clalancette, 2021-05-05:
> I'm not sure what's the exact output we're expecting from the manual work.
> I will experiment with `ament_clang_format` locally to see what I can do.

I'm going to open a draft pull request with the automatically-generated stuff in a little bit here.  We should probably fit it into that format.

ivanpauno, 2021-05-05:
I'm checking locally, `catkin_generate_changelog --all` does a really good work of generating the complete changelog, with all items under the corresponding version subtitles.

I was thinking that we might want to do the following:

- Use `catkin_generate_changelog --all` for all repos listed above.
- Cleanup the output of the generated changelog for all items that happened after Foxy.
  e.g.: for `ament_clang_format` we would need to cleanup from `0.10.0` to `0.10.5` (including both 0.10.0 and 0.10.5).
- Delete the forthcoming section (except we're also going to do a new release immediatly).
- We can keep the generated changelog items for everything that happened until Foxy without cleaning them up, or delete them.
- Use the same script that we're using for all the other repos :smiley: .

Does that sound like a good idea?

ivanpauno, 2021-05-05:
> e.g.: for ament_clang_format we would need to cleanup from 0.10.0 to 0.10.5 (including both 0.10.0 and 0.10.5).

To get that easily:

```
$ git describe --tags --contains $(git merge-base foxy master)
0.9.4
```

That's the newest tag in common with the foxy branch, i.e. that would be the one were we can stop cleaning up the changelog items (all just delete everything from there to the end of the file).

clalancette, 2021-05-05:
> Does that sound like a good idea?

Yeah, that sounds like a good idea, and it fits well into the automation that I have.  I would say that we shouldn't delete everything prior to Foxy, but we should only show the changes since Foxy in the long-form changelog.  I'll look at what it would take to my script to make it understand that.  In the meantime, I think it is worthwhile to generate the changelogs and commit them to the repositories that are missing them, regardless.

clalancette, 2021-05-05:
> I'll look at what it would take to my script to make it understand that.

OK, I was able to teach my script how to parse that fairly easily.  So I think we're good to go on this plan.

ivanpauno, 2021-05-07:
There are a few repos that I'm not sure if they should have changelogs at all:

- fastrtps
- iceoryx_*
- orocos_kdl (this one is a fork)

All those aren't maintained by the ROS 2 team (AFAIK).

clalancette, 2021-05-13:
All right, we've done all the ones we are going to do here.  So I'm going to close this out.

closed 2021-05-13 by clalancette