ros2/ros2_documentation #1466: Galactic Changelog Short Demos

cottsay, 2021-05-05:
This is a tracking ticket for ensuring that the short-form changelog entries include short demos or code snippets that so users can better understand what the entries describe and how to take advantage of them quickly.

- [ ] [Ability to specify per-logger log levels](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#ability-to-specify-per-logger-log-levels)
- [ ] [Ability to configure logging directory through environment variables](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#ability-to-configure-logging-directory-through-environment-variables)
- [ ] [https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#externally-configure-qos-at-start-up](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#externally-configure-qos-at-start-up)
- [ ] [Default RMW vendor changed to Cyclone DDS](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#default-rmw-vendor-changed-to-cyclone-dds)
- [ ] [Connext RMW changed to rmw_connextdds](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#connext-rmw-changed-to-rmw-connextdds)
- [ ] [New RMW API](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#new-rmw-api)
- [ ] [nav2](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#nav2)
- [ ] [tf2_ros Python split out of tf2_ros](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#tf2-ros-python-split-out-of-tf2-ros)
- [ ] [tf2_ros Python TransformListener uses global namespace](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#tf2-ros-python-transformlistener-uses-global-namespace)
- [ ] [rclcpp](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rclcpp)
  - [ ] [Change in spin_until_future_complete template parameters](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#change-in-spin-until-future-complete-template-parameters)
  - [ ] [Change in default /clock subscription QoS profile](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#change-in-default-clock-subscription-qos-profile)
  - [ ] [Waitable API](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#waitable-api)
  - [ ] [Change in rclcpp’s logging macros](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#change-in-rclcpp-s-logging-macros)
  - [ ] [Parameter types are now static by default](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#parameter-types-are-now-static-by-default)
  - [ ] [Add API for checking QoS profile compatibility](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#add-api-for-checking-qos-profile-compatibility)
- [ ] [rclpy](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rclpy)
  - [ ] [Removal of deprecated Node.set_parameters_callback](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#removal-of-deprecated-node-set-parameters-callback)
  - [ ] [Parameter types are now static by default](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#id1)
  - [ ] [Add API for checking QoS profile compatibility](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#id2)
- [ ] [rclcpp_action](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rclcpp-action)
  - [ ] [Action client goal response callback signature changed](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#action-client-goal-response-callback-signature-changed)
- [ ] [rosidl_typesupport_introspection_c](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rosidl-typesupport-introspection-c)
  - [ ] [API break in function that gets an element from an array](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#api-break-in-function-that-gets-an-element-from-an-array)
- [ ] [rcl_lifecycle and rclcpp_lifecycle](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rcl-lifecycle-and-rclcpp-lifecycle)
  - [ ] [RCL’s lifecycle state machine gets new init API](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rcl-s-lifecycle-state-machine-gets-new-init-api)
  - [ ] [RCL’s lifecycle state machine stores allocator instance](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rcl-s-lifecycle-state-machine-stores-allocator-instance)
  - [ ] [RCLCPP’s lifecycle node exposes option to not instantiate services](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#rclcpp-s-lifecycle-node-exposes-option-to-not-instantiate-services)

hidmic, 2021-05-14:
As I said elsewhere, while trying to craft demos for some of the entries listed here, I started to wonder about the short-form changelog itself. Some of the entries are relatively minor features or API breaks deep inside the stack. If that's the cut, many more changes listed in the complete changelog are not less worthy of being included here. 

So I went over the list. I think all the entries below are worth being mentioned (in no particular order):

- [ ] Make ament_python_install_package() install a flat Python egg (`#316 <https://github.com/ament/ament_cmake/issues/316>`__)
- [ ] Add SYSTEM keyword option to ament_target_dependencies (`#297 <https://github.com/ament/ament_cmake/issues/297>`__)
- [ ] Camera Calibration Parsers ROS2 Port (`#105 <https://github.com/ros-perception/image_common/issues/105>`__)
- [ ] Image Transport ROS2 port (`#84 <https://github.com/ros-perception/image_common/issues/84>`__)
- [ ] camera_info_manager ROS2 port (`#94 <https://github.com/ros-perception/image_common/issues/94>`__)
- [ ] Port eigen_kdl.h/cpp to ROS2 (`#311 <https://github.com/ros2/geometry2/issues/311>`__)
- [ ] Add new conditions for checking launch configuration values (`#453 <https://github.com/ros2/launch/issues/453>`__)
- [ ] Add arg_choice arg to DeclareLaunchArguments (`#483 <https://github.com/ros2/launch/issues/483>`__)
- [ ] Add respawn and respawn_delay support (`#426 <https://github.com/ros2/launch/issues/426>`__)
- [ ] Fix ComposableNode ignoring PushRosNamespace actions (`#162 <https://github.com/ros2/launch_ros/issues/162>`__)
- [ ] Substitutions in parameter files (`#168 <https://github.com/ros2/launch_ros/issues/168>`__)
- [ ] Upgrade google benchmark from v1.5.1 to v1.5.2 to include QNX patch. (`#9 <https://github.com/ament/google_benchmark_vendor/issues/9>`__)
- [ ] Update libyaml_vendor to 0.2.5. (`#37 <https://github.com/ros2/libyaml_vendor/issues/37>`__)
- [ ] Update to pybind 2.5.0 (`#2 <https://github.com/ros2/pybind11_vendor/issues/2>`__)
- [ ] Add LoadMap service (`#129 <https://github.com/ros2/common_interfaces/issues/129>`__)
- [ ] New `tango_icons_vendor <https:/github.com/ros-visualization/tango_icons_vendor/tree/master/CHANGELOG.rst>`__ package
- [ ] Quality work (API revisions, unit test, integration, performance tests, fault injection, binary mocking)
   - [ ] New `test_rmw_implementation <https:/github.com/ros2/rmw_implementation>`__ package
   - [ ] New `performance_test_fixture <https:/github.com/ros2/performance_test_fixture/>`_ package
   - [ ] Fault injection macros and functionality (plus example) (`#264 <https://github.com/ros2/rcutils/issues/264>`__)
- [ ] Unique network flows (`#880 <https://github.com/ros2/rcl/issues/880>`__)
- [ ] Add functions for waiting for publishers and subscribers (`#907 <https://github.com/ros2/rcl/issues/907>`__)
- [ ] Add generic publisher and generic subscription for serialized messages (`#1452 <https://github.com/ros2/rclcpp/issues/1452>`__)
- [ ] Allow configuring logging directory through environment variables (`#53 <https://github.com/ros2/rcl_logging/issues/53>`__)
- [ ] Allow timers to keep up the intended rate in MultiThreadedExecutor (`#1516 <https://github.com/ros2/rclcpp/issues/1516>`__)
- [ ] Add spin_all method to Executor (`#1156 <https://github.com/ros2/rclcpp/issues/1156>`__)
- [ ] Ability to configure domain_id via InitOptions. (`#1165 <https://github.com/ros2/rclcpp/issues/1165>`\_\_)
   Add ability to configure domain ID (`#596 <https://github.com/ros2/rclpy/issues/596>`__)
- [ ] Add message lost subscription event (`#232 <https://github.com/ros2/rmw/issues/232>`\_\_)
      Add message lost subscription event (`#572 <https://github.com/ros2/rclpy/issues/572>`\_\_)
      Add message lost subscription event (`#1164 <https://github.com/ros2/rclcpp/issues/1164>`__)
- [ ] Mitigate lost service responses discovery issue (`#187 <https://github.com/ros2/rmw_cyclonedds/issues/187>`__)
- [ ] Make service wait for response reader (`#390 <https://github.com/ros2/rmw_fastrtps/issues/390>`__)
- [ ] Add per-message ZSTD compression (`#418 <https://github.com/ros2/rosbag2/issues/418>`__)
- [ ] Sqlite storage double buffering (`#546 <https://github.com/ros2/rosbag2/issues/546>`__)
- [ ] Add wildcard loading to ros2 param load. (`#602 <https://github.com/ros2/ros2cli/issues/602>`__)
- [ ] Add rosparam verb load. (`#590 <https://github.com/ros2/ros2cli/issues/590>`__)
- [ ] Add --keep-alive option to 'topic pub'. (`#544 <https://github.com/ros2/ros2cli/issues/544>`__)
- [ ] Add option to echo serialized messages. (`#470 <https://github.com/ros2/ros2cli/issues/470>`__)
- [ ] PlayerClock initial implementation - Player functionally unchanged (`#689 <https://github.com/ros2/rosbag2/issues/689>`__)
- [ ] /clock publisher in Player (`#695 <https://github.com/ros2/rosbag2/issues/695>`__)
- [ ] Add to_yaml() function for C++ messages (`#527 <https://github.com/ros2/rosidl/issues/527>`__)
- [ ] Port of point_cloud2.py from ROS1 to ROS2. As seperate pkg. (`#128 <https://github.com/ros2/common_interfaces/issues/128>`__)
- [ ] TimePanel port (`#599 <https://github.com/ros2/rviz/issues/599>`__)

Some I think are super relevant for users, like the humongous quality improvements we did, the cool new `ros2bag` features, the perception package ports, the service improvements in Tier 1 RMWs, etc.

clalancette, 2021-05-23:
We've done what we are going to do here, and today is Galactic release, so closing this.

closed 2021-05-23 by clalancette