ros2/ros2_documentation #1631: Create tf2 tutorials page

kurshakuz, 2021-06-01:
I am currently in the process of migrating tf2 tutorials to ROS2, and would like to know how it would be best to update the structure of the [current](http://docs.ros.org/en/rolling/Tutorials.html) tutorials page. It will include more than 10 tutorial pages covering the same topic and would definitely feel cluttered and separated if simply added based on their difficulty levels. I will cover almost everything in the [old](http://wiki.ros.org/tf2/Tutorials)  ROS tutorials page and think that keeping similar structure will make more sense. Can anyone suggest what would be the best way to include it?

[image]

kurshakuz, 2021-06-01:
The first Introduction to tf2 tutroial is completed and is shown in: https://github.com/ros2/ros2_documentation/pull/1628

For now I added it to Beginner: CLI Tools section, but not sure if it is the correct way.

audrow, 2021-06-02:
I think that it would be good to have an intermediate page for the tf2 tutorials. Specifically, on the tutorials page, link to a page for tf2 tutorials, and from there, you can add the specific tutorials.

kurshakuz, 2021-06-02:
@audrow where should I place the link to the intermediate page in the [tutorials](http://docs.ros.org/en/rolling/Tutorials.html) page. Should it be under the Examples section?

audrow, 2021-06-02:
I think probably in the "Intermediate" section. If any of the tutorials should probably be advanced, perhaps we can have a larger discussion on how to organize them.

clalancette, 2021-06-02:
I think the whole thing should be in the intermediate section (even the introduction).  While `tf2` is an important concept in ROS, it's not really required to understand it to get started.

One other thing I'll mention here is that we can put these tutorials into `ros2_documentation` to start with, but long-term we may want to consider moving these into https://github.com/ros2/geometry2 .  We can't do that right now because we don't have a good mechanism to deploy tutorials from individual packages, but eventually we'll get there.

Aut0R3V, 2021-09-06:
Hey, this looks interesting, anything I can do here?
Thanks!

kurshakuz, 2021-09-06:
> Hey, this looks interesting, anything I can do here?
> Thanks!

Hi! I am working on this myself currently, but you can take a look at other open issues and try to resolve them. Your help will be appreciated 😉

kennywangaigraphx, 2021-09-11:
@kurshakuz Finally I finished tutorial "Using Stamped datatypes with tf2_ros::MessageFilter" last night, so can I create a PR for it, or do I need to wait for #1923 PR merged to rolling branch? You can look it in my repo. https://github.com/kennywangaigraphx/ros2_documentation

kurshakuz, 2021-09-12:
@kennywangaigraphx no, feel free to create it. We will make a review as soon as possible

aprotyas, 2021-09-14:
>General:
> - [ ] Quaterion Fundamentals

@kurshakuz Do you plan to write a page for this?

kurshakuz, 2021-09-14:
> > General:
> > 
> > * [ ]  Quaterion Fundamentals
> 
> @kurshakuz Do you plan to write a page for this?

I am currently working on other PRs, and was planning to do it next week, but if you want to do it, your help will be very appreciated! Just let me know if you are going to do it

kurshakuz, 2021-09-21:
Hello @clalancette! The biggest part of tutorials mentioned in your initial issue https://github.com/ros2/geometry2/issues/387 is covered so far. However, I would like to get your advice on the last ones. My current observation is that "Setting up your robot using tf", "Using the robot state publisher on your own robot", "Using urdf with robot_state_publisher" are already covered in one or other sense. However, they are currently spread out among different pages as shown below:

- Setting up your robot using tf
   [https://navigation.ros.org/setup_guides/transformation/setup_transforms.html](https://navigation.ros.org/setup_guides/transformation/setup_transforms.html)

- Using the robot state publisher on your own robot
   [https://navigation.ros.org/setup_guides/urdf/setup_urdf.html](https://navigation.ros.org/setup_guides/urdf/setup_urdf.html)
   [https://github.com/ros/robot_state_publisher/tree/ros2](https://github.com/ros/robot_state_publisher/tree/ros2)

- Using urdf with robot_state_publisher
  [https://docs.ros.org/en/rolling/Tutorials/URDF/Using-URDF-with-Robot-State-Publisher.html](https://docs.ros.org/en/rolling/Tutorials/URDF/Using-URDF-with-Robot-State-Publisher.html)

I am currently thinking adding those links with corresponding descriptions to the About tf2 page. What do you think about it?

clalancette, 2021-09-21:
>     * Setting up your robot using tf
>       https://navigation.ros.org/setup_guides/transformation/setup_transforms.html

For this one, I'd actually suggest we contact the Navigation team and see if they would be OK migrating that guide from the Navigation documentation into the core ROS 2 documentation.  Then they can link to it from their own documentation.  @SteveMacenski , thoughts about doing that?

>     * Using the robot state publisher on your own robot
>       https://navigation.ros.org/setup_guides/urdf/setup_urdf.html

This is another one that we could consider moving into the core ROS 2 documentation, as it is not navigation specific.  Additionally, we could definitely use some more advanced URDF and robot_state_publisher tutorials, as those are frequently asked about.  It may be outside of the scope of this issue, though.

>     * Using urdf with robot_state_publisher
>       https://docs.ros.org/en/rolling/Tutorials/URDF/Using-URDF-with-Robot-State-Publisher.html

That one is fine as-is.  If we are going to link to it from "About-Tf2", I think we should make it clear how that tutorial relates to TF2 (basically that the combination of the URDF, robot_state_publisher, and joint_state_publisher publishes transforms to TF2).

SteveMacenski, 2021-09-27:
These are pretty specific pages to Nav2. I can see they would have general utility, but keep in mind our goal in these pages are to teach people to set up the transforms / sensors for **Nav2**, not in general (e.g. not in cars, manipulators, or other robot types). We don't bring up the conventions in naming or styling for those other forms so it would probably need some additions to be more generic for the ROS 2 documentation.

The code this references is in the Nav2 tutorials github repo, how would you suggest dealing with that?