ros2/ros2_documentation #1678: Ros2 tutorials, creating workspace colcon fail - turtlesim

Tjaart-Hobby, 2021-06-13:
Hi.
I am new to ros and ros2 so installed ros2 following [https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html](url)
I am going through the tutorials and got up to [https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html](url), so when I run the "colcom build" command, it fails:

_Starting >>> turtlesim
--- stderr: turtlesim                         
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
 ...... bla.bla.bla...
---
Failed   <<< turtlesim [0.37s, exited with code 1]

Summary: 0 packages finished [0.56s]
  1 package failed: turtlesim
  1 package had stderr output: turtlesim
tk@tk-Aspire-5820T:~/dev_ws$_ 

I was able to sort out errors up to now but google failed me with this one.
Also attempted: [https://docs.ros.org/en/galactic/Tutorials/Colcon-Tutorial.html#colcon](url) from where I installed colcon, but still errors out.
my ros2 installation of Galactic and Foxy are all good,
I do source when opening a new terminal (note source command different to tutorials command but all ros2 commands work)
expected result according to tutorial:

_`Starting >>> turtlesim
Finished <<< turtlesim [5.49s]

Summary: 1 package finished [5.58s]`_

How do i create a works space with colcon? Whats wrong?
note: the bold print of first half of error not intentional, cant seem to un-bold it, sorry.

Tjaart-Hobby, 2021-06-14:
never mind found [https://stackoverflow.com/questions/58913364/could-not-find-openssl-try-to-set-the-path-to-openssl-root-folder](url) so just needed to install " sudo apt install libssl-dev " build sucsessfull

closed 2021-06-14 by Tjaart-Hobby

miercat0424, 2022-04-14:
this saved my life

miercat0424, 2022-04-14:
> 

> never mind found [https://stackoverflow.com/questions/58913364/could-not-find-openssl-try-to-set-the-path-to-openssl-root-folder](url) so just needed to install " sudo apt install libssl-dev " build sucsessfull

this method saved my life