ros2/ros2_documentation #1735: Add tutorial/guide for creating a C/C++ package with Python code using ament_cmake_python

christophebedard, 2021-07-12:
I've seen a couple questions about `ament_cmake_python`. Someone thought it was a build type (like `ament_cmake`), someone else wanted to include Python code in a CMake (`ament_cmake`) package. I don't think they thoroughly searched through the docs, but if they did, they wouldn't find much.

It's not really mentioned much other than here: https://docs.ros.org/en/rolling/Concepts/About-Build-System.html#the-ament-cmake-repository. But it simply links to the source code, which itself doesn't have any examples: https://github.com/ament/ament_cmake/tree/master/ament_cmake_python

One can grep through a ROS 2 core workspace and find a few examples, like `rcutils`, but it's not used much. https://github.com/ros2/rcutils/blob/712122f8554bb62dff78cefe8611cff403c62d2e/CMakeLists.txt#L19

The proposed tutorial/guide would show the required file structure:

```
.
└── pkg_a
    ├── CMakeLists.txt
    ├── package.xml
    └── pkg_a
        ├── __init__.py
        └── my_script.py
```

show the new dependency:

```xml
<buildtool_depend>ament_cmake_python</buildtool_depend>
```

show the CMake code:

```cmake
find_package(ament_cmake_python REQUIRED)
# ...
ament_python_install_package(${PROJECT_NAME})
```

and maybe show how it can be used/`import`ed by another package just to wrap up the example:

```python
# in pkg_b
from pkg_a.my_script import my_function
```

If this sounds like a good idea, where should it go? Guides?

clalancette, 2021-07-19:
Yeah, we should definitely have documentation about it.  That said, it's important not to emphasis it too much; pure Python packages really should be using the `ament_python` type, not `ament_cmake_python`.  So I'd be fine with adding some additional documentation about it, but being really clear that this should only be used in circumstances where you cannot use the normal Python disttools mechanisms.

christophebedard, 2021-07-19:
> That said, it's important not to emphasis it too much; pure Python packages really should be using the `ament_python` type, not `ament_cmake_python`. So I'd be fine with adding some additional [...]

Sounds good, thanks for the feedback!

Aut0R3V, 2021-09-05:
@christophebedard any progress on this? If not then I'd like to try this out

christophebedard, 2021-09-08:
@Aut0R3V I'll write up something for this soon!

PR #1943 (2021-09-14): Add ament_cmake_python user documentation
Closes #1735

There's a "basics" section and nothing else because I wasn't sure if I should mention the other CMake macros, like:

* `ament_get_python_install_dir`: https://github.com/ament/ament_cmake/blob/099871343a8e0b6cee2e90cdc815dd61eaf5abf3/ament_cmake_python/cmake/ament_get_python_install_dir.cmake
* `ament_python_install_module`: https://github.com/ament/ament_cmake/blob/099871343a8e0b6cee2e90cdc815dd61eaf5abf3/ament_cmake_python/cmake/ament_python_install_module.cmake

The last one isn't used anywhere from what I can tell.

If #1385 gets pick [... cut]
closed 2021-09-14 by christophebedard