ros2/ros2_documentation #1736: Add a page for where things are and what to put where

audrow, 2021-07-13:
Currently, I have trouble knowing where ROS 2 related information should live. It would be nice to have a page in the documentation that explains what service/organization is used for what.

This could include the following
- Github, including the different organizations
- Docs, including when to look at tutorials, guides, and concepts
- Discourse
- ROS Answers
- REP
- Build/CI
- Dockerhub

Aut0R3V, 2021-09-05:
@audrow any progress on this? If not so, then I'd like to give it a shot

clalancette, 2021-09-06:
We could definitely use help here.  My suggestion for this is to put it in the "Contributing" section of the documentation, probably as a new page under there.

knmcguire, 2026-01-11:
Actually... [the landing page of docs.ros.org](https://docs.ros.org/) fulfills a lot of these as well. Is that the page that solves the problem this issue is created for? Seems like the most logical spot tbh

It does misses github/dockerhub/ci but perhaps that is something we could add.