ros2/ros2_documentation #1738: Prerequisites missing from "Installing ROS 2 on Windows" page?

eren-rudy, 2021-07-13:
I had a bit of an adventure getting my Windows 10 install of galactic working, and I think its possible this might have been due to some missing instructions on the [Installing ROS 2 on Windows](http://docs.ros.org/en/galactic/Installation/Windows-Install-Binary.html) page.

I followed the instructions for installing galactic on Windows 10 and everything went well (other than a warning about a missing Connext DDS file under Environment Setup, which is fine because I didn't install Connext). However, when I got to the examples provided to test the installation, the program exited with an error (unfortunately, I was shortsighted and didn't save the stack trace).

I jumped to the [Building ROS 2 on Windows](http://docs.ros.org/en/galactic/Installation/Windows-Development-Setup.html) instructions, and noticed that there were a few prerequisites under [Additional prerequisites](http://docs.ros.org/en/galactic/Installation/Windows-Development-Setup.html#additional-prerequisites) that are **not** listed on the [Installing ROS 2 on Windows](http://docs.ros.org/en/galactic/Installation/Windows-Install-Binary.html) instructions. I installed these additional prerequisites (actually, only the [Chocolatey](http://docs.ros.org/en/galactic/Installation/Windows-Development-Setup.html#install-additional-prerequisites-from-chocolatey) and [Python prerequisites](http://docs.ros.org/en/galactic/Installation/Windows-Development-Setup.html#install-python-prerequisites), separated by a reboot as prompted after the Chocolatey installations), after which the ROS 2 talker/listener examples began working.

Is there any possibility that these prerequisites belong in the installation instructions? I did try something else (I tried -- unsuccessfully, I had issues when building from source with rosdep -- to perform a manual installation of Eclipse Cyclone DDS, as suggested on a ROS forum [here](https://answers.ros.org/question/379164/rti-connext-dds-environment-script-not-found-ros2-on-windows/)) but this proved to be fruitless. I mention this to address the chance that the 'fix' described above might be a false positive, and something else might have happened that I failed to account for. If it seems possible that those two "Additional prerequisites" from the "Building ROS 2 on Windows" instructions may actually be necessary for the regular installation instructions, I would be happy to try to reproduce my initial issue more systematically.

Apologies if this was a bit long-winded, and thanks for your help!

clalancette, 2021-07-13:
> I jumped to the [Building ROS 2 on Windows](http://docs.ros.org/en/galactic/Installation/Windows-Development-Setup.html) instructions, and noticed that there were a few prerequisites under [...]

Theoretically, none of the prerequisites in [Additional prerequisites](http://docs.ros.org/en/galactic/Installation/Windows-Development-Setup.html#additional-prerequisites) are necessary in order to run, only to build from source and/or run tests.  That said, it is certainly possible that something is missing from the [Installing ROS 2 on Windows](http://docs.ros.org/en/galactic/Installation/Windows-Install-Binary.html) page.  It's just not clear to me what is missing without a stack trace.

But pinging @cottsay , who dealt with Windows installation most recently.  He may have some additional feedback here.

laercio-barbosa, 2023-03-26:
I have installed ROS2 Humble binary following [these](https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html) instructions. I have had no problem at all. Chocolaty and Python are there as prerequisites.

@eren-rudy 
Have you tried to install a newer version since then? 

@clalancette 
Maybe this issue can be closed and issues counter be decreased 😄

clalancette, 2023-03-26:
Yes, given the age of this issue, I'm going to close it out.  If you are still having a problem, please open a new one.

closed 2023-03-26 by clalancette