ros2/ros2_documentation #1776: Remove old "Using tf2 with ROS 2" tutorial?

kurshakuz, 2021-07-24:
The first tutorial page on tf2 called "Using tf2 with ROS 2" looks unnecessary now.

The information presented in the https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/tf2.rst was covered in more details in new tutorials on tf2:

1. The information on using ``static_transform_publisher`` as in
https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/tf2.rst#using-tf2-with-ros-2
is presented in
https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Tf2/Writing-A-Tf2-Static-Broadcaster-Py.rst#the-proper-way-to-publish-static-transforms

2. The information on using ``tf2_echo`` as in 
https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/tf2.rst#receiving-transform-data
is presented in
https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Tf2/Writing-A-Tf2-Broadcaster-Py.rst#3-build-and-run

Should we removed or update it somehow?

clalancette, 2021-08-02:
If the information contained in it is available elsewhere, I'm all for removing the old one.

kurshakuz, 2021-08-13:
Removed in https://github.com/ros2/ros2_documentation/pull/1807

closed 2021-08-13 by kurshakuz