ros2/ros2_documentation #1778: Ros2 Foxy Turtle Tutorial qt missing?

Markland01, 2021-07-25:
Had issues with getting the turtle sim tutorial to work. Found a work around for this issue:

> qt.qpa.plugin: Could not find the Qt platform plugin "windows" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

[https://answers.ros.org/question/354707/qtqpaplugin-could-not-find-the-qt-platform-plugin-windows-in/](url) and I feel you should know and or add it to the tutorial?

audrow, 2021-08-04:
Hi @Markland01, thanks for pointing this out. Feel free to make a PR adding this note to the tutorial. If you do, you can @ me as a reviewer.

Aut0R3V, 2021-09-05:
When I open the link provided by @Markland01, the webpage doesn't show anything. Suggestions?

aprotyas, 2021-09-05:
> When I open the link provided by @Markland01, the webpage doesn't show anything. Suggestions?

For some reason that link itself is text to a separate URL? Just copy the text: https://answers.ros.org/question/354707/qtqpaplugin-could-not-find-the-qt-platform-plugin-windows-in/

Aut0R3V, 2021-09-06:
Oh, that worked! Apologies for bugging you about such an insignificant problem.
I think this would go here - https://docs.ros.org/en/foxy/Installation.html
Or please recommend where you think would be the ideal place for it, Thanks!

audrow, 2021-09-07:
No worries. I fixed the link in the description to point to the correct URL.

febu1300, 2022-04-25:
I think this issue is fixed but its not yet deployed.
https://github.com/ros2/ros2_documentation/blob/rolling/source/Installation/_Windows-Install-Prerequisites.rst#install-qt5

clalancette, 2024-10-11:
Sine I think this was fixed long ago, I'm going to close this out.  If you are still having issues, please feel free to reopen.

closed 2024-10-11 by clalancette