ros2/ros2_documentation #1787: Windows 10 Installation - add a link to Microsoft IoT RosOnWindows instructions

jonathanTIE, 2021-07-31:
There are so many dependencies and it's a nightmare for windows 10. I tried for hours to solve each error that popped, but it never ended. Then I found and followed the instructions from the Microsoft IoT team, just replacing foxy with galactic and it worked in a few minutes. 

The instructions are here : https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html

Until the installation become simpler, the documentation should suggest the users who want to get started quickly to use these instructions first.

christophebedard, 2021-08-05:
Those installation instructions are definitely simpler, but it does things a bit differently and it can lead to issues. For example, here `ament_cmake_pytest` couldn't find pytest: https://github.com/ament/ament_cmake/issues/274. Those instructions and the resulting setup/binaries aren't tested as well as the current instructions/setup/binaries. Personally, based on that, I'm not sure we should suggest that users use those instructions *first*.

Of course, with a bit of help we could fix the issues that come up.

The Dashing instructions actually do mention those instructions as an alternative at the end (https://docs.ros.org/en/dashing/Installation/Windows-Install-Binary.html#alternative-ros-2-build-installation-from-aka-ms-ros), so maybe we should bring that back?

audrow, 2021-08-05:
> The Dashing instructions actually do mention those instructions as an alternative at the end (https://docs.ros.org/en/dashing/Installation/Windows-Install-Binary.html#alternative-ros-2-build-installation-from-aka-ms-ros), so maybe we should bring that back?

Seems reasonable to me.

jonathanTIE, 2021-08-07:
I'm still trying to make things work with these binaries, I encountered issues with the released foxy version and galactic pre-release (not available yet) with rqt and nav2 (some dependencies are not updated to the correct version when using ros2 doctor), but the foxy pre-release from Microsoft seems to work, I just needed the Rqt dependencies. The current foxy documentation doesn't use the "--pre" arg, so maybe we should only update the foxy documentation to include it like the getting started from ms website and wait for the official support of galactic ?

traversaro, 2021-08-08:
Mentioning @ooeygui that could be interested in this discussion.

closed 2024-10-11 by clalancette