ros2/ros2_documentation #1864: Documentation regarding making ROS2 cross-platform-OS

PrasadNR, 2021-08-23:
Currently ROS2 appears to work only on Focal Fossa of Ubuntu 20.04 LTS. I was thinking about using this with Gazebo. http://gazebosim.org/tutorials?tut=ros2_overview

So, there are three main problems:

1. I am not able to see any documentation as to why ROS2 works only on Ubuntu 20.04 LTS and not other Linux distributions in general.
2. I am not able to see any documentation as to how I can contribute to ROS2 so that it works on many usual versions of Windows and Linux (and potentially others can help with Mac based ROS2).
3. I am not able to see any documentation regarding the OS level interaction between Ubuntu 20.04 and ROS2; Or why it is so "tightly bound" with Ubuntu 20.04 LTS (and how I can de-couple it if needed).

PrasadNR, 2021-08-23:
Also, another aspect: I am not really sure why `sudo` is needed and what I can do to avoid using `sudo` or root access. I was thinking of having ROS2 as a software package and not really something requiring root access (something like Webots; But, definitely not exactly like Webots).

gavanderhoorn, 2021-08-23:
Are you referring to the *amount* of information you can find for OS other than Ubuntu, or really support in the sense you can build and/or install ROS 2 on other OS?

I'm asking as I could understand the first, but the second option would confuse me a bit:

[image]

Could you clarify what gave you this impression:

> I am not able to see any documentation as to why ROS2 works only on Ubuntu 20.04 LTS

?

christophebedard, 2021-08-23:
> 1\. I am not able to see any documentation as to why ROS2 works only on Ubuntu 20.04 LTS and not other Linux distributions in general.
>
> 3\. I am not able to see any documentation regarding the OS level interaction between Ubuntu 20.04 and ROS2; Or why it is so "tightly bound" with Ubuntu 20.04 LTS (and how I can de-couple it if needed).

See this Discourse thread: https://discourse.ros.org/t/why-does-the-ros2-distro-need-to-be-pinned-to-a-specific-ubuntu-lts/17435. I think the "documentation" for this (at least a short version of the long answer in the Discourse thread) is here: https://www.ros.org/reps/rep-2000.html#targeted-platforms

PrasadNR, 2021-08-24:
@gavanderhoorn 

> Are you referring to the amount of information you can find for OS other than Ubuntu, or really support in the sense you can build and/or install ROS 2 on other OS?

Yeah. Being able to install ROS 2 on other OS (it is ok if I really cannot build it natively). By other OS, I mean, different versions of Ubuntu and different Linux variants (not a specific version of a particular variant; for example, https://docs.ros.org/en/galactic/Installation/RHEL-Install-RPMs.html is supported on RHEL 8 and not other versions of RHEL).

>Could you clarify what gave you this impression:
> I am not able to see any documentation as to why ROS2 works only on Ubuntu 20.04 LTS _and not other Linux distributions in general._

I have a Chromebook which does not support Ubuntu 20.04 via crouton. I am thinking about some other libraries and other hardwares which might not work properly if I have to use Ubuntu 20.04 (NVIDIA GPU drivers being one problem I can think of). 

@christophebedard https://discourse.ros.org/t/why-does-the-ros2-distro-need-to-be-pinned-to-a-specific-ubuntu-lts/17435 This link is really helpful. Thank you!

PrasadNR, 2021-08-25:
Then, after reading through this, it does look like there are some efforts to have Docker containers for those and then, there are some efforts towards conda and pip packages. I do not think it is appropriate to expect documentation regarding conda and pip packages when they have not been developed. I am closing this for now. But, I would certainly see a greater influence of ROS2 if this is supported via some stand-alone packages (with or without docker). Then, this "issue" is more of the development than of the documentation.

closed 2021-08-25 by PrasadNR

traversaro, 2021-08-25:
> Then, after reading through this, it does look like there are some efforts to have Docker containers for those and then, there are some efforts towards conda and pip packages.

Just FYI, some experimental instructions on how to install ROS2 Galactic via conda (so without root access and also on non-Ubuntu distro) can be found in https://github.com/RoboStack/ros-galactic , The package are still experimental, but feel free to try them if you like and report any issue you may find in https://github.com/RoboStack/ros-galactic, thanks!