ros2/ros2_documentation #1881: Following TransformBroadcaster tutorial results in a memory leak

ijsmallman, 2021-08-25:
We recently spent a frustrating day trying to get to the bottom of a bug that had a symptoms of slowly increasing latency in updates to the TF tree and increasing node memory usage. Eventually we figured out that someone had copied code from the tutorial [https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Tf2/Writing-A-Tf2-Broadcaster-Py.rst](url).

The pose subscriber callback has the following lines:

```python
    def handle_turtle_pose(self, msg):
        # Initialize the transform broadcaster
        br = TransformBroadcaster(self)
```

I believe this results in an ever increasing number of publishers being added to the node `__publishers` list. The `TransformBroadcaster` should be setup in the node `__init__` method.

ahcorde, 2021-08-25:
I created an issue https://github.com/ros/geometry_tutorials/issues/55 in the `geometry_tutorial` repo. Thank you for reporting.

kurshakuz, 2021-08-25:
Updated in https://github.com/ros2/ros2_documentation/pull/1882. Thanks for noticing that!

ahcorde, 2021-08-25:
The `geometry_tutorial` repository is updated as well as the documentation.

Feel free to reopen it if the issue is still there

closed 2021-08-25 by ahcorde