ros2/ros2_documentation #1907: ROS2 foxy install FAILS tries to install hirsute

JHPHELAN, 2021-09-06:
In the section "And then add the repository to your sources list:" the command
`echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null
`
creates a /etc/apt/sources.list.d/ros2.list file:
`deb [arch=arm64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu hirsute  main
`
instead of foxy.  If corrected to foxy, the command
`sudo apt update`
gives the errors:
`Ign:1 http://packages.ros.org/ros2/ubuntu foxy InRelease
Err:2 http://packages.ros.org/ros2/ubuntu foxy Release
  404  Not Found [IP: 64.50.236.52 80]
Reading package lists... Done
E: The repository 'http://packages.ros.org/ros2/ubuntu foxy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`
This was reported previously this year and attributed to an outdated key and supposedly fixed.  But it's back and seems to be new within the last week as I installed foxy ok within the last month.

gavanderhoorn, 2021-09-06:
The output of `lsb_release -c` is the *codename* of the Ubuntu / Debian release, not the name of the ROS 2 version.

The error is expected, as `foxy` is not a valid Ubuntu release codename.

clalancette, 2021-09-06:
> The error is expected, as `foxy` is not a valid Ubuntu release codename.

That's exactly right.  At the end of the day, the line in `/etc/apt/sources.list.d/ros2.list` is saying "fetch ROS packages from this URL that target a specific Ubuntu version".  In this case, it looks like you are running on Ubuntu 21.04, which is why "hirsute" is added.

With all of that said, we don't ship Foxy pre-packaged for Ubuntu 21.04; we only ship it packaged for Ubuntu 20.04.  So this setup won't work out-of-the-box.  Your options are:

1.  Do your development in an Ubuntu 20.04 container (this is what I do)
2. Build ROS 2 from source (though we don't officially target hirsute with any ROS distribution, so you may have to fix some things along the way).
3. Downgrade your machine to Ubuntu 20.04.

Regardless, this doesn't look like a bug, so I'm going to close this.  If you still need help with this, please ask a question on https://answers.ros.org.

closed 2021-09-06 by clalancette