ros2/ros2_documentation #2027: package 'turtle_tf2_py' not found

algorithmlover2016, 2021-10-16:
according to the demo of https://docs.ros.org/en/galactic/Tutorials/Tf2/Introduction-To-Tf2.html#intrototf2
install cmd is: sudo apt-get install ros-galactic-turtle-tf2-py ros-galactic-tf2-tools
and I also run the cmd: pip3 install transforms3d
however when I run the demo cmd: ros2 launch turtle_tf2_py turtle_tf2_demo.launch.py, there is an error:
Package 'turtle_tf2_py' not found: "package 'turtle_tf2_py' not found, searching:[*]
I find the searching path is not included the following.
I run the "dpkg -L ros-galactic-turtle-tf2-py ros-galactic-tf2-tools", it returns:
`/.
/opt
/opt/ros
/opt/ros/galactic
/opt/ros/galactic/lib
/opt/ros/galactic/lib/python3.8
/opt/ros/galactic/lib/python3.8/site-packages
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/init.py
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/pycache
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/pycache/init.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/pycache/dynamic_frame_tf2_broadcaster.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/pycache/fixed_frame_tf2_broadcaster.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/pycache/static_turtle_tf2_broadcaster.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/pycache/turtle_tf2_broadcaster.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/pycache/turtle_tf2_listener.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/dynamic_frame_tf2_broadcaster.py
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/fixed_frame_tf2_broadcaster.py
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/static_turtle_tf2_broadcaster.py
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/turtle_tf2_broadcaster.py
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py/turtle_tf2_listener.py
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py-0.3.0.egg-info
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py-0.3.0.egg-info/PKG-INFO
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py-0.3.0.egg-info/dependency_links.txt
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py-0.3.0.egg-info/entry_points.txt
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py-0.3.0.egg-info/requires.txt
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py-0.3.0.egg-info/top_level.txt
/opt/ros/galactic/lib/python3.8/site-packages/turtle_tf2_py-0.3.0.egg-info/zip-safe
/opt/ros/galactic/lib/turtle_tf2_py
/opt/ros/galactic/lib/turtle_tf2_py/dynamic_frame_tf2_broadcaster
/opt/ros/galactic/lib/turtle_tf2_py/fixed_frame_tf2_broadcaster
/opt/ros/galactic/lib/turtle_tf2_py/static_turtle_tf2_broadcaster
/opt/ros/galactic/lib/turtle_tf2_py/turtle_tf2_broadcaster
/opt/ros/galactic/lib/turtle_tf2_py/turtle_tf2_listener
/opt/ros/galactic/share
/opt/ros/galactic/share/ament_index
/opt/ros/galactic/share/ament_index/resource_index
/opt/ros/galactic/share/ament_index/resource_index/packages
/opt/ros/galactic/share/ament_index/resource_index/packages/turtle_tf2_py
/opt/ros/galactic/share/turtle_tf2_py
/opt/ros/galactic/share/turtle_tf2_py/launch
/opt/ros/galactic/share/turtle_tf2_py/launch/turtle_tf2_demo.launch.py
/opt/ros/galactic/share/turtle_tf2_py/launch/turtle_tf2_dynamic_frame_demo.launch.py
/opt/ros/galactic/share/turtle_tf2_py/launch/turtle_tf2_fixed_frame_demo.launch.py
/opt/ros/galactic/share/turtle_tf2_py/package.xml
/opt/ros/galactic/share/turtle_tf2_py/rviz
/opt/ros/galactic/share/turtle_tf2_py/rviz/turtle_rviz.rviz
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-galactic-turtle-tf2-py
/usr/share/doc/ros-galactic-turtle-tf2-py/changelog.Debian.gz

/.
/opt
/opt/ros
/opt/ros/galactic
/opt/ros/galactic/lib
/opt/ros/galactic/lib/python3.8
/opt/ros/galactic/lib/python3.8/site-packages
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools/init.py
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools/pycache
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools/pycache/init.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools/pycache/view_frames.cpython-38.pyc
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools/view_frames.py
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools-0.17.2.egg-info
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools-0.17.2.egg-info/PKG-INFO
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools-0.17.2.egg-info/dependency_links.txt
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools-0.17.2.egg-info/entry_points.txt
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools-0.17.2.egg-info/requires.txt
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools-0.17.2.egg-info/top_level.txt
/opt/ros/galactic/lib/python3.8/site-packages/tf2_tools-0.17.2.egg-info/zip-safe
/opt/ros/galactic/lib/tf2_tools
/opt/ros/galactic/lib/tf2_tools/view_frames
/opt/ros/galactic/share
/opt/ros/galactic/share/ament_index
/opt/ros/galactic/share/ament_index/resource_index
/opt/ros/galactic/share/ament_index/resource_index/packages
/opt/ros/galactic/share/ament_index/resource_index/packages/tf2_tools
/opt/ros/galactic/share/tf2_tools
/opt/ros/galactic/share/tf2_tools/package.xml
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-galactic-tf2-tools
/usr/share/doc/ros-galactic-tf2-tools/changelog.Debian.gz`

algorithmlover2016, 2021-10-16:
I know the reason,  I build ros2 from source and I just source the ~/ros2_galactic/install/setup.bash. I forget to source /opt/ros/galactic/setup.bash.
so the solution is:
. ~/ros2_galactic/install/setup.bash
.  /opt/ros/galactic/setup.bash

christophebedard, 2021-10-16:
Great! I'll close this issue.

closed 2021-10-16 by christophebedard