ros2/ros2_documentation #2112: Visual Studio 2019 has now been replaced with Visual Studio 2022

christophebedard, 2021-11-11:
We've been using Visual Studio 2019 for a while now. I'm following the build instructions. By following the link here, it's a bit hard & annoying to actually find Visual Studio 2019: https://docs.ros.org/en/rolling/Installation/Windows-Development-Setup.html#install-visual-studio. The website tries real hard to get you to install the 2022 release. I had to login with a Microsoft account to get Visual Studio Community 2019.

So I'm wondering:

* should we switch to Visual Studio 2022? Do we expect it to just work?
* or should we mention having to find Visual Studio 2019, at least in the meantime?

clalancette, 2021-11-11:
>     * should we switch to Visual Studio 2022? Do we expect it to just work?

Generally, that has not been our experience in the past.  Likely we'll have to do some code changes.  Even if we do get it to work, we are also going to need to fix up https://ci.ros2.org to use the new version.  I think that is something we *should* do before Humble, but it will take some dedicated time.

> * or should we mention having to find Visual Studio 2019, at least in the meantime?

Yeah, I think we'll have to do this for now.  A direct link to that version would be best, if we can find it.

christophebedard, 2021-11-11:
> Yeah, I think we'll have to do this for now. A direct link to that version would be best, if we can find it.

This is the link I got: https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=Community&rel=16&src=myvs&utm_medium=microsoft&utm_source=my.visualstudio.com&utm_campaign=download&utm_content=vs+community+2019. Seems like you don't actually need to be logged in or anything.

aprotyas, 2021-11-11:
> This is the link I got: https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=Community&rel=16&src=myvs&utm_medium=microsoft&utm_source=my.visualstudio.com&utm_campaign=download&utm_content=vs+community+2019.

This link works for me too.

PR #2123 (2021-11-15): Use direct link to Visual Studio 2019
Closes #2112
closed 2021-11-15 by christophebedard