ros2/ros2_documentation #2124: "Building a Visual Robot Model with URDF from Scratch" doesn't work on Foxy

christophebedard, 2021-11-15:
See https://answers.ros.org/question/390892/how-to-upgrade-launch_ros/

The urdf_tutorial repo/package has a launch file that uses `ParameterValue` from `launch_ros`: https://github.com/ros/urdf_tutorial/blob/09a5289c6cd7a6222a5538bcf77a09a73e251f76/launch/display.launch.py#L23.

However, it doesn't exist in Foxy (well, it exists, but as a type). See https://github.com/ros2/launch_ros/pull/137

Since the urdf_tutorial repo only has a `ros2` branch, a new `foxy` branch might need to be created to get a fix for Foxy (unless a temporary universal fix is applied to the `ros2` branch until Foxy is EOL or something).

cc @DLu

christophebedard, 2021-11-16:
> However, it doesn't exist in Foxy (well, it exists, but as a type). See [ros2/launch_ros#137](https://github.com/ros2/launch_ros/pull/137)

This was just backported to Foxy in https://github.com/ros2/launch_ros/pull/265, so this issue should be fixed whenever the next Foxy patch is released.

DLu, 2021-12-06:
Can we close this as of the [last sync](https://discourse.ros.org/t/new-packages-for-foxy-fitzroy-2021-12-02/23333)?

closed 2021-12-06 by christophebedard