ros2/ros2_documentation #2193: Tutorial "Using parameters in a class (Python)" has a small mistake (default value =? empty string)?

MihailV1989, 2021-12-31:
In Chapter 2.1 there is the following confusing statement:

> Then, we set the parameter ‘my_parameter’ **_back_** to the default string value ‘world’.

And in chapter 3 of the tutorial "Using parameters in a class (Python)" there is the following statement:

> Except the first message where the parameter had a default value (**_an empty string_**), the terminal should return the following message every 2 seconds:

`[INFO] [parameter_node]: Hello world!`

But the parameter is first declared like this:
`self.declare_parameter('my_parameter', 'world')`

And then there is the option to add a ParameterDescriptor in chapter 2.1.1, there the parameter is declared like this:

> self.declare_parameter('my_parameter',
>                                'default value for my_parameter',
>                                my_parameter_descriptor)

In both cases the default value is not an empty string, isn't it? To me as ROS newbie the parameter usage was already confusing enough. I couldn't find an opened issue about this problem so I thought I should submit a new.

It would also help to have a background explanation why you have to always create a new reference and set the parameter type in order to just change the parameter value like this:

>     my_new_param = rclpy.parameter.Parameter(
>         'my_parameter',
>         rclpy.Parameter.Type.STRING,
>         'world'
>     )
>     all_new_parameters = [my_new_param]
>     self.set_parameters(all_new_parameters)

The name "my_new_param" for the reference is also kind of confusing. It isn't an actual new parameter that is being created, isn't it? Is there more to it?

clalancette, 2022-01-04:
Thanks for the report.

> > Then, we set the parameter ‘my_parameter’ **_back_** to the default string value ‘world’.

Sorry, its unclear to me what is confusing about this statement.  That is literally what that line of code is doing; setting the `my_parameter` parameter back to `world`.  The user may have changed it in the meantime, but this causes the code to only print the "new" value once, and then reset it back to `world` for the next iteration.

> > Except the first message where the parameter had a default value (**_an empty string_**), the terminal should return the following message every 2 seconds:

That's a bug in the tutorial.  In earlier versions of ROS 2, that would be empty, but that is no longer the case.  I'll submit a PR to fix that.

> > self.declare_parameter('my_parameter',
> > 'default value for my_parameter',
> > my_parameter_descriptor)

Yeah, the second argument there should be 'world'.  I'll also fix that.

> From the rclpy API it seems like giving a parameter type is not mandatory:
> https://docs.ros2.org/foxy/api/rclpy/api/parameters.html
> so the tutorial would be more helpful showing that the type is not needed, which should be the most common use case, isn't it? Especially as it seems that by default a parameter cannot change type, unless a `ParameterDescriptor(dynamic_typing=True)` is passed upon declaration.

While you are right that it is not mandatory, the type is the second argument of the function.  So if you are using keyword arguments, you can skip it, but if you are using just regular arguments you cannot.  This feels like a subjective argument, so I'm going to leave it as-is.

> Then the name "my_new_param" for the reference is also kind of confusing. It isn't an actual new parameter that is being created, isn't it?

No, it's a new Parameter object.  We are then using that Parameter object to change the value of `my_parameter`.

> Is there more to it? In the first parameters tutorial it's mentioned that the parameters functionality is built off of ROS 2 services, so maybe the function rclpy.parameter.Parameter() creates a message that is being sent as request to change the parameter?

So there are 2 different ways to change parameters in ROS 2.  If you have a reference Node object (i.e. you are in the same process), then you can just use the APIs for manipulating the parameters directly.  If you are in a separate process, then you can use the parameter services as provided by the node.  In this case, we are in the same process so we don't use that latter functionality.

> P.S. Second question: why do we import ParameterType:
> `from rcl_interfaces.msg import ParameterType`
> when we don't use it anywhere?

Leftover from previous versions of the tutorial where we did use it.  I'll remove this as well.

> Third question: How ROS knows what is the type of the parameter, when when we didn't set it? For a string like in the tutorial is probably not needed: `self.declare_parameter('my_parameter', 'world')` but for a number? Or when we don't set a default value? It would be really nice to have such very common user scenarios covered in the tutorial.

It infers it from the default value that is passed.  If no default value is passed, then you need to pass the type explicitly when calling `declare_parameter`.

> It would be really nice to have such very common user scenarios covered in the tutorial.

To be totally clear, the beginner tutorials are not meant to cover every possible scenario.  They are meant to show beginners one recommended way to do things.  If we want to have more in-depth coverage of parameter, then I think a separate tutorial or a How-To Guide might be more appropriate for that.

clalancette, 2022-01-04:
See https://github.com/ros2/ros2_documentation/pull/2202 , which fixes the most glaring issues.

MihailV1989, 2022-01-04:
Thanks for the clarifications! 

> Sorry, its unclear to me what is confusing about this statement. That is literally what that line of code is doing; setting the my_parameter parameter back to world. The user may have changed it in [...]

The confusing thing to me was that in the code the parameter was not changed from it's default value. But yes, I didn't considered, that it could have been changed by the user.

> To be totally clear, the beginner tutorials are not meant to cover every possible scenario. They are meant to show beginners one recommended way to do things. If we want to have more in-depth [...]

For me it would be enough to add your answer in chapter 2.1:

> It infers it from the default value that is passed. If no default value is passed, then you need to pass the type explicitly when calling declare_parameter.

and also adding that you have to use ParameterDescriptor() for that. 

Having an additional tutorial for the ParameterDescriptor() would be definitely nice. Getting used to using messages for setting up different function needs practice.

clalancette, 2022-01-04:
> The confusing thing to me was that in the code the parameter was not changed from it's default value. But yes, I didn't considered, that it could have been changed by the user.

That's fair.  I've added one new sentence about this in #2202.

> > 
> 
> For me it would be enough to add your answer in chapter 2.1:
> 
> > It infers it from the default value that is passed. If no default value is passed, then you need to pass the type explicitly when calling declare_parameter.
> 
> and also adding that you have to use ParameterDescriptor() for that.

I've added one new sentence in #2202 about that as well.  Note that you don't need to use a ParameterDescriptor; you can just pass the type in the second argument.

clalancette, 2022-01-20:
Closing now that #2202 is in and fixes most of the issues.  If you have more problems, please feel free to open up a new issue.  Thanks for the report.

closed 2022-01-20 by clalancette