ros2/ros2_documentation #2239: Add example of C++ logging implementation

cadmus-to, 2022-01-23:
(Referenced in #2026)

> I'm not sure if this should be a separate issue, but it is within the scope of this one.
> 
> Under Logging Usage for C++ at the Concept page (https://docs.ros.org/en/rolling/Concepts/About-Logging.html), there probably should be a C++ implementation demo similar to the Python's one. Probably something like:
> 
> - https://github.com/ros2/demos/tree/master/logging_demo ?

Yeah just thought it'd be good to put a link for that there, in a similar fashion to the Python demo.

kurshakuz, 2022-01-26:
Having similar content present under different languages makes a lot of sense to me. However, the package you mentioned seems to serve a little bit different purpose and is not as intuitive as [Python example](https://github.com/ros2/examples/blob/master/rclpy/services/minimal_client/examples_rclpy_minimal_client/client.py). I have found similar [implementation in C++](https://github.com/ros2/examples/blob/master/rclcpp/services/minimal_client/main.cpp) which implements the same example and could fit better.

clalancette, 2023-06-20:
We've significantly redone the logging documentation since this issue was opened, and I now think that http://docs.ros.org/en/rolling/Concepts/About-Logging.html does a much better job of explaining what is available.

Based on that, I'm going to close this out.  But feel free to reopen if you still think this is an issue.

closed 2023-06-20 by clalancette