ros2/ros2_documentation #2275: Default Galactic installed DDS(es)

shamlian, 2022-02-04:
I'm not sure the note in the [Galactic] DDS Implementations](https://github.com/ros2/ros2_documentation/blob/galactic/source/Installation/DDS-Implementations.rst) page is correct; I am only finding Cyclone installed after following the [official directions](https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Binary.html) in Ubuntu 20.04 (in fact, these docs seem to imply that only Cyclone _should_ be installed). Which is the intent? For my use case, it would be convenient if `ros-galactic-rmw-fastrtps-cpp` were part of `ros-galactic-ros-base` but that's easy for me to fix with a slightly longer apt invocation. :)

clalancette, 2022-02-04:
You are absolutely right; that page is out-of-date, and should say that CycloneDDS is the default.  We updated most of the places that said that, but apparently we missed that one.  Would you mind submitting a PR to update it?

shamlian, 2022-02-04:
Not at all! Just wanted to make sure I was correct before trying to fix something that wasn't broken.

shamlian, 2022-02-04:
Created #2276

kurshakuz, 2022-02-08:
Thank you for the contribution! Closing the issue now.

closed 2022-02-08 by kurshakuz