ros2/ros2_documentation #2309: Links to rclcpp and rclpy API in rolling distro documentation refer to foxy related API

ehiker, 2022-02-16:
Hello,

I found the rolling distro documentation (at least [internal interfaces](https://docs.ros.org/en/rolling/Concepts/About-Internal-Interfaces.html?highlight=interface
) and [client interfaces](https://docs.ros.org/en/rolling/Concepts/About-Client-Interfaces.html)) refers to API documentation related to foxy distro ([rclcpp API](https://docs.ros2.org/foxy/api/rclcpp/index.html), [rclpy API](https://docs.ros2.org/foxy/api/rclpy/) and [rmw API](https://docs.ros2.org/foxy/api/rmw/index.html)).

I think foxy API documentation is outdated in relation to rolling. I could not find rolling API documentation (something like [https://docs.ros2.org/rolling/api/rclcpp/index.html](https://docs.ros2.org/rolling/api/rclcpp/index.html)). I think the links should be updated to galactic because it is more related documentation than the foxy one.

aprotyas, 2022-02-17:
Thanks for catching that. I agree that each distribution's page should link to the API documentation corresponding to that distribution.

I forget why, but rolling API docs can be found at `https://docs.ros.org/en/ros2_packages/rolling/api/[package-name]/index.html`.

For some context:
Looking at the [roadmap](https://docs.ros.org/en/rolling/Roadmap.html#humble-roadmap) for the next distribution, Humble, there's a documentation related item: "Documentation: Upload C++ API documentation for core packages to docs site". If I remember correctly, the API documentation link structure may change in that process. I'm assuming this would also have been taken care of as part of that effort.

clalancette, 2023-09-18:
This was actually fixed a while ago, so I'm going to close this one out.

closed 2023-09-18 by clalancette