ros2/ros2_documentation #2333: Documentation Error - Efficient Intra Process Communication

chedanix, 2022-02-26:
The website is https://docs.ros.org/en/foxy/Tutorials/Intra-Process-Communication.html?highlight=intra
The Github file is located at `source/Tutorials/Intra-Process-Communication.rst`

Under the section **Pipeline with two image viewers**, the two image viewers do not show the correct information. When I perform the same command `ros2 run intra_process_demo image_pipeline_with_two_image_view`, all six addresses are **exactly the same**. However, in the current documentation, one of the six addresses is different, as shown below.
[image]

The documentation points out the address being different. But the explanation then says the two image viewers should share the same image resource. This is contradicting because if the same image resource is shared, all six addresses should be the same, as shown below.
[image]

My system:
Ubuntu 20.04
ROS Galactic

ivanpauno, 2022-03-17:
I think the wording is correct but the image was never updated, right?
The same seems to happen in [this section](https://docs.ros.org/en/foxy/Tutorials/Intra-Process-Communication.html?highlight=intra#pipeline-with-interprocess-viewer).

> I would like to become a contributor. Please let me fix it if this is indeed an error.

Sounds good!
Feel free to tag me in the PR to ask for a review.

roncapat, 2022-04-30:
I think this is partially correct... but:
- first two addresses of both images shall be always the same.
- one of the two images shall also show the third address equal to the first two
- the other one should be different

If I'm not wrong, only 5 addresses shall be equal then. 

Here is my screenshot:
[image]

I can make a PR, but please confirm me first my reasoning is ok.

ivanpauno, 2022-05-02:
> I can make a PR, but please confirm me first my reasoning is ok.

Yes, that's fine.

> The same seems to happen in [this section](https://docs.ros.org/en/foxy/Tutorials/Intra-Process-Communication.html?highlight=intra#pipeline-with-interprocess-viewer).

I think this one could also be updated.
Again, 5 pointers should be the same and the image view node in a separate process should've a different pointer.

roncapat, 2022-05-08:
Correct. However, the tutorial tells the user to stop the viewers to check for addresses, while actual ROS2 intra-process and demo nodes implementation will always show a stable address for every one of the six shown, so no more need to stop the viewers and tell the user to do so neiter. Shall I rewrite the affected paragraphs also?

roncapat, 2022-05-08:
Another question: shall I push to rolling? I tested the above assumptions on ROS2 Foxy, so from Foxy onwards it should be ok.

roncapat, 2022-05-08:
Here is the two screenshots, ready for inclusion

https://imgur.com/a/N8dz122

I can do the push when someone tells me the right branch :)

fujitatomoya, 2022-05-09:
> shall I push to rolling?

yes, we can fix the rolling and then we can do backport. that is usual procedure.

ijnek, 2022-08-14:
The **Efficient Intra Process Communication** guide seems heavily outdated, it mentions that unique_ptr is the only way to achieve zero-copy intra process comms, but there's been major improvements and shared_ptrs can be used too.

I believe https://design.ros2.org/articles/intraprocess_communications.html is packed with the most recent intra process design and we should rewrite this guide based off that.

---

@chedanix:
> I would like to become a contributor. Please let me fix it if this is indeed an error.

If there is quite a bit of work, are you still willing to do the work or are you happy for someone else to make the changes?

clalancette, 2022-08-15:
> The **Efficient Intra Process Communication** guide seems heavily outdated, it mentions that unique_ptr is the only way to achieve zero-copy intra process comms, but there's been major improvements and shared_ptrs can be used too.

From an end-user point-of-view, I don't think it is heavily outdated (just mildly outdated).  That is, while the implementation details of how we deliver data between publishers and subscribers has changed significantly, I don't think that it makes a large change on that API.  That said, I could be wrong so looking over the page and fixing it up with the latest information is definitely appreciated.

> I believe https://design.ros2.org/articles/intraprocess_communications.html is packed with the most recent intra process design and we should rewrite this guide based off that.

Just to be doubly clear, I don't think we should pull all of those details in here.  This is a tutorial, so is meant to show the user how to use this stuff, not exactly how it is implemented.  We should definitely have a link to the design article in there, and should make sure the content is up-to-date, but I wouldn't put details of the ring buffers (for instance) in that tutorial.

ijnek, 2022-08-16:
@clalancette I believe the code itself is incorrect, but is missing some important points that have changed due to PRs that have been merged since the article was first written. Although it may be just be personal preference, the way the code is written also doesn't seem to follow the current best practices of writing a node.

> Just to be doubly clear, I don't think we should pull all of those details in here. This is a tutorial, so is meant to show the user how to use this stuff, not exactly how it is implemented. We [...]

Yes, I don't mean to put the details in here. We can add a link to it, but more importantly come up with demos that cover the capabilities of the intra process comms that have changed over time.

I've raised a new issue to discuss improvements: #2955

fujitatomoya, 2025-11-04:
@mini-1235 i believe that we need to update the documentation for this.

wjwwood, 2025-11-13:
> While using sensor_msgs::msg::Image::ConstSharedPtr will result in zero copy, therefore all 6 pointers address should be the same

This doesn't sound right to me. The whole point of the demo (in my opinion as the original author) was to show that you can deliver a unique pointer to exactly one subscriber and that a duplicate would be made for other subscribers asking for a read-write unique pointer to the message. I need to do some archaeology here to figure out what when wrong.

mini-1235, 2025-11-13:
> The whole point of the demo (in my opinion as the original author) was to show that you can deliver a unique pointer to exactly one subscriber and that a duplicate would be made for other subscribers asking for a read-write unique pointer to the message.

Sorry, I missed that it is requesting for a read-write pointer. I think what said was true before https://github.com/ros2/demos/pull/532 is merged, should we consider updating the subscription callback signature instead of changing the docs?

wjwwood, 2025-11-17:
@mini-1235 I think the code change that caused the docs to be wrong was this (ancient) change: https://github.com/ros2/demos/commit/456b5bac8f527141ed4e5af73168b3d31adb9b22

And the docs were never updated, but in my opinion the code change was the thing that shouldn't have changed, as the docs reflect the point of this part of the demo. So please give me a bit of time to fix both the code and the docs.

wjwwood, 2025-11-21:
Sorry, I meant to close the pull request, I reopened the issue until the pull requests I opened are merged.

closed 2025-11-24 by ahcorde