ros2/ros2_documentation #2401: URDF Doc link pointing to ros 1

vkuehn, 2022-04-14:
Somehow the documentation is mixed up for ros2 with ros1 links. E.g.

https://github.com/ros2/ros2_documentation/blob/foxy/source/Tutorials/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.rst

points to the ros1 tutorial with the effect that new ones like me get mixed up installations. 

urdf_tutoria -> [urdf_tutorial](https://index.ros.org/p/urdf_tutorial)

maybe should be
urdf_tutorial -> https://docs.ros.org/en/foxy/Tutorials/URDF/Using-URDF-with-Robot-State-Publisher.html

a link check during the doc build process would be greatly appreciated

kurshakuz, 2022-04-16:
Hello @vkuehn, I have just looked into your issue and can not exactly understand what you mean by saying "mixed up for ros2 with ros1 links". The link you mentioned - [urdf_tutorial ](https://index.ros.org/p/urdf_tutorial) points out to the ROS Index website which is not a documentation website, but rather a website that stores indexed list of all packages available for both ROS 1 and ROS 2. You can learn more about it [here](https://docs.ros.org/en/foxy/Docs-Guide.html#ros-index-section).

At which step exactly you got mixed up installations? Did you install the _urdf_tutorial_ package according to your ROS 2 distro?

vkuehn, 2022-04-16:
@kurshakuz  thanks for the link

from time to time I try to start with ros again on a certain topic.Usually I fail to be productive as just following links brings me to ros 1 documentation. As an Example the URDF stuff for ros2.

https://github.com/ros2/ros2_documentation/blob/foxy/source/Tutorials/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.rst

has ....If you installed [urdf_tutorial](https://index.ros.org/p/urdf_tutorial) binaries, 
has ... API Docs -> 404
and is the galaxy documentation

I would have expected that the Link for a urdf_tutorial goes not to the libraries but the foxy documentation mentioned above.

than on https://docs.ros.org/en/foxy/Tutorials/URDF/Using-URDF-with-Robot-State-Publisher.html there is a link to rviz2 which ends up on the https://index.ros.org/r/rviz/ which is galaxy libraries but should galaxy, foxy etc not be rviz2 ?

I am a bit confused by that and is only one example. A clarification how to read documentation and start with ros2 only without following all the baby steps and having ros2 knowledge would be probably good.

kurshakuz, 2022-04-16:
@vkuehn yeah, I now see your confusion!

>  has ... API Docs -> 404

Unfortunately, a lot of ROS 2 packages are still missing the API Docs pages, so you will have to go to the source code or readme files to learn how to work with them.

> I would have expected that the Link for a urdf_tutorial goes not to the libraries but the foxy documentation mentioned above.

Your concern is reasonable, but the link should point to the API Documentation, not the tutorial page that uses that package. Links in the [readme file ](https://github.com/ros/urdf_tutorial) are already pointing to those tutorials, so I think it should be leaved this way.

> than on https://docs.ros.org/en/foxy/Tutorials/URDF/Using-URDF-with-Robot-State-Publisher.html there is a link to rviz2 which ends up on the https://index.ros.org/r/rviz/ which is galaxy libraries but should galaxy, foxy etc not be rviz2 ?

Regarding this one you are absolutely correct! The link is pointing to the repository of the RViz package, where both RViz and RViz2 source codes are hosted. This link should rather be changed to https://index.ros.org/p/rviz2/github-ros2-rviz/

>  A clarification how to read documentation and start with ros2 only without following all the baby steps and having ros2 knowledge would be probably good.

That sounds like a reasonable concern. Do you have any ideas what can be done to improve that reading experience?

vkuehn, 2022-04-16:
Just let a rooky like me Work through a single topic 😉
A documentation for a fresh ros2 Installation in wach topic would probably clarify a lot even If that means a lot rerepeating.

kurshakuz, 2022-04-16:
No problem! I opened a PR solving this confusion. I will close that issue after it is merged.

kurshakuz, 2022-04-18:
Solved in https://github.com/ros2/ros2_documentation/pull/2402

closed 2022-04-18 by kurshakuz

vkuehn, 2022-04-18:
@kurshakuz oh sorry understood to late, just realised you did already the work. Sorry and Thanks a lot