ros2/ros2_documentation #2442: ROS 2 docs organization makes information difficult to navigate

esthersweon, 2022-04-28:
ROS 2 docs are difficult to navigate for several reasons:
- Hundreds of pages in a mostly flat hierarchy (e.g. "Tutorials" – not enough subfolders)
- Related items that aren’t grouped together meaningfully (i.e. info on a particular concept like ROS params is scattered across multiple sections and pages, Roadmap and Feature status/ideas are all separate pages, etc.)
- Content is often duplicated across multiple pages in different sections (i.e. Tutorials vs. How-To Guides cover similar content in different ways)
- Grounding info that helps users get the lay of the land is also spread across multiple pages (home page + Docs Guide page
- General copy editing – Large blocks of text, obsolete or deprecated information, not enough lists or subsections to help the user break up the huge amount of information, etc.

**Short-term goal:**
- Rethink the hierarchy, ordering, and organization of the existing content to be easier to navigate and digest
- Copy edit these existing pages for conciseness and clarity

**Long-term goal:**
- Remove obsolete or irrelevant content
- Consolidate duplicated content across pages
- Move less relevant content off to more appropriate sites in the ROS ecosystem
- Consolidate sections that contain duplicated content (e.g. Tutorials and How-To Guides are not different enough to warrant an artificial separation)
- Determine what content can be moved off into a “Miscellaneous” folder

**Proposed 1st phase for making docs easier to navigate:** 
- Tackle the introductory pages, Installation section, and Tutorials section.
- Group the "Roadmap", "Feature status", and "Feature ideas" pages into one section

**Proposed tasks for this phase:**
- Home and Docs Guide pages should be consolidated into one page and heavily copy edited for conciseness and clarity
- [docs.ros.org](http://docs.ros.org/) and home page should be explicitly added to the sidebar for easy discovery (not immediately obvious that the home icon or ROS logo gets you back to the home page)
- Add explanatory subfolders where needed and revise page names to be less verbose / repetitive

**Tentative outline:**

- Introduction (current home page)
- Installation
   - Introduction (should be a combination of the current “Installing the latest ROS 2 development” + “Maintaining a source checkout…” + “Alternative Installation Sources for Testing” pages)
     - Linux (subfolder)
       - Ubuntu Linux
       - Fedora Linux
     - Debian Packages (subfolder)
     - macOS (subfolder)
     - RHEL (subfolder)
     - DDS (subfolder)
- Tutorials
     - Configuring your environment
     - Understanding the ROS 2 graph
         - Prerequisite: Setup with turtlesim + rqt
         - Nodes, topics, and services (consolidate these pages)
         - Parameters
             - Using parameters in a class
                 - C++
                 - Python
             - Monitor changes
                 - C++
         - Actions
             - Creating an action
             - Writing an action server and client
                 - C++
                 - Python
     - Viewing logs
     - Launch
         - Introduction (Current “Introducing ROS 2 launch”)
         - Rest of pages in Launch Tutorials
     - Recording and playback
         - Recording a bag from your own node
             - C++
             - Python
     - Creating workspace _ first ROS 2 package
     - Writing a publisher and subscriber
         - C++
         - Python
     - Writing a service and client
         - C++
         - Python
     - ROS 2 Security
         - Introduction
         - Understanding the keystone
         - Across machines
         - Network traffic
         - Access Controls
     - [[OTHER PAGES AS IS]]
- [[OTHER SECTIONS AS IS]]
- Current status and roadmap
     - Features status
     - Roadmap (will include feature ideas)  

Would be happy to open a PR for this, if this outline and direction look good.

clalancette, 2022-04-28:
In general, a reorganization of the information to make it easier to navigate is long overdue.  So thanks for starting the conversation.

I have specific quibbles with your outline, however, particularly around the Installation pages (the "development setup" should probably be the last thing in the list, not in the Introduction), and around the Tutorials (the general idea is we want to steer people towards the "Beginner" tutorials as a group to start with, and then direct them to specific tutorials for other things).  That said, I'm happy to review pull requests towards something like the above, and I can point out my specific quibbles in the individual pull requests.

I have a couple of general comments when opening pull requests to move towards the following:

1.  Make sure to open, small, bite-sized pull requests to make the changes you want.  If you open one giant one, it's going to be impossible to review and much more likely to introduce regressions.
2.  If you move the URLs of pages around, make sure to put redirects in them.  An example of doing this is https://github.com/ros2/ros2_documentation/blame/rolling/source/How-To-Guides.rst#L1 , where we make sure to redirect from https://docs.ros.org/en/rolling/Guides.html (the old name) to https://docs.ros.org/en/rolling/How-To-Guides.html .  This goes for removing pages as well; if you remove a page, make sure to add a `redirect-from` on the page that contains the de-duplicated content.

But besides that, I welcome these changes and look forward to your help!

clalancette, 2022-06-27:
@esthersweon We've now merged in a lot of your great work to help fix this:

* https://github.com/ros2/ros2_documentation/pull/2453
* https://github.com/ros2/ros2_documentation/pull/2455
* https://github.com/ros2/ros2_documentation/pull/2461
* https://github.com/ros2/ros2_documentation/pull/2465

Looking at the state of the current site, I think the last two things would be to somehow organize "How-to Guides" as well as "Concepts".  Both of those are a little harder as there are no clear lines around which to organize, but maybe we could do something.

@esthersweon Are you interested in trying out those two?  Or are you happy enough with the site as it is to leave it be and to close this issue?

(as I've said before, thanks so much for your work here)

clalancette, 2023-06-20:
So we did end up revamping the Tutorials, thanks to the work of @esthersweon .  We also have an open PR to reorganize the Conceptual documents in https://github.com/ros2/ros2_documentation/pull/3732 .

Beyond that, I don't have a concrete plan for more reorganization.  I'm going to close this one out, and we can always consider rearranging more in the future.

closed 2023-06-20 by clalancette