ros2/ros2_documentation #2443: Tutorial about how to write a component

ivanpauno, 2022-04-28:
Opening ticket based on discussion here https://github.com/ros-perception/perception_pcl/pull/359#discussion_r842029457.

It seems that we have a [tutorial](https://docs.ros.org/en/rolling/Tutorials/Composition.html) about how to add components to a container and interact with them, and we have a [concept](https://docs.ros.org/en/rolling/Concepts/About-Composition.html?highlight=rclcpp_components_register_node#writing-a-component) page for components, but we don't seem to have any tutorial of how to write a them.

clalancette, 2022-04-28:
Yep, agreed this would be great to have.  I'll put it on the Humble board for now.

SteveMacenski, 2022-05-13:
[image]

(its Friday, we can have fun on Fridays, right?)