ros2/ros2_documentation #2444: Use permalink instead of copy&paste the source code in documents

fujitatomoya, 2022-04-28:
It would be really nice and good for maintenance if we can link the source code as permalink instead of doing `copy&paste` the related source code in this repository.

example: https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Intra-Process-Communication.rst `code-block`

fujitatomoya, 2022-04-28:
question came up on https://github.com/ros2/ros2_documentation/pull/2441#pullrequestreview-956935799

kurshakuz, 2022-05-02:
There are already a couple of examples that do use the source code directly in the rst files like:

1. https://github.com/ros2/ros2_documentation/rolling/source/Tutorials/Actions/Writing-a-Cpp-Action-Server-Client.rst
2. https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Actions/Writing-a-Py-Action-Server-Client.rst

Look for `.. literalinclude:: ` directive.

Unfortunately, those examples require to host code snippets locally. I am not sure if it is possible to directly fetch code samples from other repositories.

gavanderhoorn, 2022-07-06:
Would depending on something like [sphinxext-remoteliteralinclude](https://pypi.org/project/sphinxext-remoteliteralinclude) be acceptable for this (Github repository: [wpilibsuite/sphinxext-remoteliteralinclude](https://github.com/wpilibsuite/sphinxext-remoteliteralinclude))?

It seems to address the exact issue reported here: be able to retrieve code from a remote location, instead of duplicating it next to the `.rst` using `literalinclude`.

fujitatomoya, 2024-04-03:
I will go ahead to close this issue, it is gonna be tough to keep the consumable and stable documentation instead of permlink. feel free to reopen this, if someone gets interested.

closed 2024-04-03 by fujitatomoya