ros2/ros2_documentation #2446: ROS 2 Humble needed release notes

clalancette, 2022-04-29:
Earlier this week I generated the [complete changelog](https://docs.ros.org/en/rolling/Releases/Humble-Hawksbill-Complete-Changelog.html) for Humble using a script.  I then spent several hours reading that changelog, looking for things to add to the [short-form changelog](https://docs.ros.org/en/rolling/Releases/Release-Humble-Hawksbill.html) (I think my eyes are now permanently glazed over).  While doing this, what I was looking for were user-facing APIs or features, not bug-fixes.  What I came up with is the following list of changes that would ideally be documented for Humble.  Note that there are probably both false-positives (there are things on the list that don't need to be documented) as well as false-negatives (I may have missed a new feature here or there).  Still, I think this represents a decent overall look at everything new in Humble.

1. [X] Type Adaptation (https://github.com/ros2/rclcpp/pull/1557, https://github.com/ros2/examples/pull/300)
2. [ ] Overlay include changes
3. [ ] ament_cmake_gen_version_h package (https://github.com/ament/ament_cmake/pull/198)
4. [ ] FindPython3 instead of FindPythonInterp? (https://github.com/ament/ament_cmake/pull/355)
5. [ ] ament_cmake_copyright file exclusion (https://github.com/ament/ament_lint/pull/328)
6. [ ] ament_generate_version_header (https://github.com/ament/ament_cmake/pull/377) (possibly obsoletes ament_cmake_gen_version_h
7. [ ] ament_cmake_uncrustify file exclusion (https://github.com/ament/ament_lint/pull/330)
8. [ ] SPDX identifiers for package licenses (https://github.com/ament/ament_lint/pull/315)
9. [ ] Deprecation of void shared_ptr<MessageT> subscription callback (https://github.com/ros2/rclcpp/pull/1713, https://github.com/ros2/examples/pull/323, https://github.com/ros-visualization/interactive_markers/pull/84, and others)
10. [ ] Type Adaptation with intra-process comms in same context for passing opaque handles
11. [ ] wait_for_all_acked (https://github.com/ros2/rclcpp/pull/1662, https://github.com/ros2/examples/pull/316, https://github.com/ros2/rmw/pull/296, and others)
12. [ ] rclcpp::WaitSet (https://github.com/ros2/examples/pull/315)
13. [ ] rclpy SIGINT/SIGTERM handling (https://github.com/ros2/rclpy/pull/844, https://github.com/ros2/examples/pull/330, and others)
14. [ ] Deprecation of tf2_geometry_msgs.h (https://github.com/ros2/geometry2/pull/418)
15. [ ] Launch boolean substitutions (https://github.com/ros2/launch/pull/598)
16. [ ] Launch environment variable scoping (https://github.com/ros2/launch/pull/601)
17. [ ] Launch evaluate math in python (https://github.com/ros2/launch/pull/557)
18. [ ] Launch_testing pytest (https://github.com/ros2/launch/pull/528)
19. [ ] Launch Executable description (https://github.com/ros2/launch/pull/454)
20. [ ] Launch match target action (https://github.com/ros2/launch/pull/540)
21. [ ] Launch AppendEnvironmentVariable (https://github.com/ros2/launch/pull/543)
22. [ ] Launch ClearLaunchConfigurations (https://github.com/ros2/launch/pull/515)
23. [ ] Launch_ros parameter substitution (https://github.com/ros2/launch_ros/pull/297)
24. [ ] Launch_ros Composable parameters from file (https://github.com/ros2/launch_ros/pull/281)
25. [ ] Launch_ros alternate parameter file configuration (https://github.com/ros2/launch_ros/pull/259)
26. [ ] Launch_ros SetParametersFromFile (https://github.com/ros2/launch_ros/pull/260)
27. [ ] Launch_ros change push-ros-namespace to push_ros_namespace (https://github.com/ros2/launch_ros/pull/250)
28. [X] Launch_ros new ros_arguments for Node (https://github.com/ros2/launch_ros/pull/249)
29. [ ] Launch_ros RosTimer action (https://github.com/ros2/launch_ros/pull/244)
30. [ ] Launch_ros XML composable containers (https://github.com/ros2/launch_ros/pull/235)
31. [ ] launch_testing Remove deprecated ready_fn (https://github.com/ros2/launch/pull/589)
32. [X] rosidl_get_typesupport_target (https://github.com/ros2/rosidl/pull/606)
33. [ ] content-filtered-topics (https://github.com/ros2/rcl/pull/894, https://github.com/ros2/design/pull/282, https://github.com/ros2/rmw/pull/302, https://github.com/ros2/rmw_connextdds/pull/68, https://github.com/ros2/rmw_fastrtps/pull/513, and others)
34. [ ] Events Executor support (https://github.com/ros2/rcl/pull/839, https://github.com/ros2/rmw/pull/286, https://github.com/ros2/rmw_fastrtps/pull/468, and others)
35. [ ] Client/service QoS getters (https://github.com/ros2/rcl/pull/941, https://github.com/ros2/rmw_fastrtps/pull/560, and others)
36. [ ] Allow users to disable loaned messages (https://github.com/ros2/rcl/pull/949)
37. [ ] rclcpp Removal of APIs deprecated during Galactic (https://github.com/ros2/rclcpp/pull/1913)
38. [ ] Deprecate CamelCase rclcpp_components methods (https://github.com/ros2/rclcpp/pull/1716)
39. [ ] Allow selection of executor in rclcpp_components_register_node (https://github.com/ros2/rclcpp/pull/1898)
40. [ ] Auto-transition lifecycle entities (https://github.com/ros2/rclcpp/pull/1863)
41. [ ] Deprecation of get_env.h in favor of env.h (https://github.com/ros2/rcutils/pull/340)
42. [ ] Client/service QoS getters (https://github.com/ros2/rmw/pull/314)
43. [ ] Sequence numbers in message_info (https://github.com/ros2/rmw/pull/318)
44. [X] Certificate Revocation Lists (https://github.com/ros2/rmw_dds_common/pull/52)
45. [ ] Changed default Fast-RTPS publication mode to synchronous (https://github.com/ros2/rmw_fastrtps/pull/571)
46. [ ] robot_state_publisher add frame_prefix parameter (https://github.com/ros/robot_state_publisher/pull/159)
47. [ ] robot_state_publisher subscription SensorDataQoS (https://github.com/ros/robot_state_publisher/pull/179)
48. [X] robot_state_publisher always publish fixed frames to /tf_static (https://github.com/ros/robot_state_publisher/pull/158)
49. [ ] ros2launch launch-prefix (https://github.com/ros2/launch_ros/pull/254)
50. [ ] ros2launch launch-prefix-filter (https://github.com/ros2/launch_ros/pull/261)
51. [X] ros2param command-line argument interpretation (https://github.com/ros2/ros2cli/pull/684)
52. [X] ros2param deprecate command-line arguments (https://github.com/ros2/ros2cli/pull/638)
53. [ ] ros2cli always line buffer (https://github.com/ros2/ros2cli/pull/659)
54. [ ] ros2pkg use local git config instead of global (https://github.com/ros2/ros2cli/pull/693)
55. [ ] ros2pkg automatically add LICENSE files for known licenses (https://github.com/ros2/ros2cli/pull/650)
56. [ ] ros2topic output data in YAML "block" style (https://github.com/ros2/ros2cli/pull/698)
57. [ ] ros2topic echo print once (https://github.com/ros2/ros2cli/pull/695)
58. [ ] ros2topic regex filter (https://github.com/ros2/ros2cli/pull/654)
59. [ ] ros2topic pub make QoS Reliable and Transient-local (https://github.com/ros2/ros2cli/pull/653)
60. [X] ros2topic pub add option to wait for N matching subscriptions (https://github.com/ros2/ros2cli/pull/642)
61. [ ] ros2topic pub/echo qos autodetection (https://github.com/ros2/ros2cli/pull/613)
62. [ ] ros2topic echo deprecate --lost-messages (https://github.com/ros2/ros2cli/pull/633)
63. [ ] rosbag2 bag rewrite (https://github.com/ros2/rosbag2/pull/921)
64. [ ] rosbag2 start time play option (https://github.com/ros2/rosbag2/issues/931)
65. [ ] rosbag2 ignore-leaf-topics (topics without a publisher) (https://github.com/ros2/rosbag2/pull/925)
66. [ ] rosbag2 add ability to start paused (https://github.com/ros2/rosbag2/pull/905, https://github.com/ros2/rosbag2/pull/904)
67. [ ] rosbag2 add ability to use keyboard to pause/resume/play-next (https://github.com/ros2/rosbag2/pull/847)
68. [ ] rosbag2 add snapshot mode (only start writing once service is called (https://github.com/ros2/rosbag2/pull/851)
69. [ ] rosbag2 add delay option (only start playing after delay) (https://github.com/ros2/rosbag2/pull/789)
70. [ ] rosbag2 seek interface (https://github.com/ros2/rosbag2/pull/836)
71. [ ] rosbag2 burst mode (play as fast as possible) (https://github.com/ros2/rosbag2/pull/977)
72. [ ] rosidl_generator_c implement copy and equality operators for C messages (https://github.com/ros2/rosidl/pull/650, https://github.com/ros2/rosidl/pull/648)
73. [ ] Update to Connext 6
74. [ ] sensor_msgs Add YUV420 and YUV444 to image encodings (https://github.com/ros2/common_interfaces/pull/172)
75. [ ] sensor_msgs add equidistant distortion model (https://github.com/ros2/common_interfaces/pull/160)
76. [ ] shape_msgs Add PRISM type to SolidPrimitive (https://github.com/ros2/common_interfaces/pull/167)
[... cut ...]
82. [X] rviz2 Display triangle textures with gradients (https://github.com/ros2/rviz/pull/719)
83. [ ] rviz2 Show mass properties in robot model (https://github.com/ros2/rviz/pull/714)
84. [ ] rviz2 Add ability to show YUV images in rviz (https://github.com/ros2/rviz/pull/701)
85. [ ] rclpy Add LifecycleNodes (https://github.com/ros2/rclpy/pull/865)
86. [ ] rclpy Add Clock.sleep_until and Clock.sleep_for (https://github.com/ros2/rclpy/pull/858, https://github.com/ros2/rclpy/pull/864)
87. [ ] rclcpp Add Clock::sleep_until (https://github.com/ros2/rclcpp/pull/1814)

jacobperron, 2022-05-10:
I can help add release notes related to launch, specifically, I think these ones are worth mentioning:

-  Launch environment variable scoping (https://github.com/ros2/launch/pull/601)
- Launch_testing pytest (https://github.com/ros2/launch/pull/528)
- Launch_ros XML composable containers (https://github.com/ros2/launch_ros/pull/235)

ivanpauno, 2022-05-10:
I can do at least a few:

* wait_for_all_acked (https://github.com/ros2/rclcpp/pull/1662, https://github.com/ros2/examples/pull/316, https://github.com/ros2/rmw/pull/296, and others)
* rclpy Add LifecycleNodes (https://github.com/ros2/rclpy/pull/865)

I will check the list and claim a few more later.

jacobperron, 2022-05-10:
I think we can omit https://github.com/ros2/launch_ros/pull/259, since it is a bugfix that should probably be backported.

jacobperron, 2022-05-10:
I've checked off boxes for items being added in https://github.com/ros2/ros2_documentation/pull/2486

I've added a strike-through for those items that I didn't think worth mentioning with the following rationale:

- launch Executable description (https://github.com/ros2/launch/pull/454)
  - This is mainly an implementation refactor to allow for other kinds of executables in the future. AFAIK it shouldn't affect current users of ExecuteProcess, Node, or node component related actions.
- launch_ros alternate parameter file configuration (https://github.com/ros2/launch_ros/pull/259)
  - This is a bugfix that should be backported to Galactic.
- launch_ros change push-ros-namespace to push_ros_namespace (https://github.com/ros2/launch_ros/pull/250)
  - Doesn't change anything for users. Worth noting in a future release if we decide to deprecate the old style.
- launch_testing Remove deprecated ready_fn (https://github.com/ros2/launch/pull/589)
  - The removed API has been deprecated since Foxy. IMO, it seems overly verbose to highlight this in the release notes since users should already expect it to be gone.

clalancette, 2022-05-24:
We've done the release notes we are going to do, so closing this out.

closed 2022-05-24 by clalancette