ros2/ros2_documentation #2467: Tf2 tutorials require pip packages

clalancette, 2022-05-04:
When following the tutorial at https://docs.ros.org/en/humble/Tutorials/Tf2/Writing-A-Tf2-Static-Broadcaster-Py.html , everything worked until actually attempting to run the code with:

```
ros2 run learning_tf2_py static_turtle_tf2_broadcaster mystaticturtle 0 0 1 0 0 0
```

When doing that, I got an error:
```
Installing the transforms3d library by hand required. Please run
	sudo pip3 install transforms3d
```

While I get that using tf_transformations is convenient here, it is not the out-of-the-box experience that I think we should have for our users.  We should find another way to transform an euler angle to a quaternion.

kurshakuz, 2022-05-05:
Agreed, the absence of such an essential functionality also bothers me. One approach that I see is to make David's package `transforms3d` independent. Some good effort put into the `tf_transformations` API and ... PR :D, and making fork of functions from the `transforms3d` could be a viable solution. I believe the tf.transformation appeared as a fork of https://github.com/cgohlke/transformations in the first place. Moreover, the current state of `transforms3d` seems stable in terms of functionality, as there was no commits made for the last year. It may be a good time to make a "fork" from it and make updates on demand. What do you think about it?

kurshakuz, 2022-08-25:
Solved by https://github.com/ros2/ros2_documentation/pull/2977

closed 2022-08-30 by ahcorde