ros2/ros2_documentation #2532: Composition node How-To Needs LoadComposableNodes / Container demo

SteveMacenski, 2022-05-12:
https://docs.ros.org/en/foxy/How-To-Guides/Launching-composable-nodes.html

Needs an example of using `LoadComposableNodes` to load the nodes into the container as well, so external nodes in other launch files can be used (or using `LoadComposableNodes`'s condition API for conditional launch in pre-Humble distributions).

Just a small gap in documentation and this also isn't shown in the demos/tutorials/examples repositories, so unless you knew about it, there's no reference to `LoadComposableNodes` anywhere but source code.

SteveMacenski, 2022-05-13:
In progress

SteveMacenski, 2022-05-13:
PR https://github.com/ros2/ros2_documentation/pull/2559

clalancette, 2023-06-20:
Fixed by #2559, so closing.

closed 2023-06-20 by clalancette