ros2/ros2_documentation #2558: Problem with vcs import src < ros2.repos

cjychenjiayi, 2022-05-13:
I don't know why I failed so many times on this step.
I tried to change the website in repos file but nothing worked.
[image]
this is what shown on my terminal(part of the feedback)
I can visit all those website on my  browser but it seems unaccessible for the code.
Looking forward to help, thx

fujitatomoya, 2022-05-13:
this is related to `git` and `ssh`, maybe you can check https://stackoverflow.com/questions/18356502/github-failed-to-connect-to-github-443-windows-failed-to-connect-to-github.

clalancette, 2022-06-02:
No response in over two weeks, so closing.  If you are still having problems, please open a question on https://answers.ros.org.

closed 2022-06-02 by clalancette